txoof/PaperPi #14: Quality of life improvements in install scripts

txoof, 2022-02-15:
> I'd like to make a few more quality of life improvements for the end user like adding a way to see a list of compatible screens and maybe adding at least one plugin to the base ini file (even if it's disabled by default).
> _Originally posted by @aaronr8684 in https://github.com/txoof/PaperPi/issues/7#issuecomment-1040565871_

- [ ] add more default plutins
- [ ] provide list of compatible screens during install process

txoof, 2022-02-15:
```
#!/usr/bin/env python3
import sys
from epdlib.Screen import *
sys.path.append('./paperpi')
list_compatible_modules()
```
that codesnip will produce a list of all the modules. It's a little verbose, but it could do the job:

```
NN. Board        Supported:
---------------------------
00. epd1in02     False
    Issues:
     * AttributeError: module does not support `EPD.display()`
01. epd1in54     True
02. epd1in54_V2  True
03. epd1in54b    True
04. epd1in54b_V2 True
05. epd1in54c    True
06. epd2in13     True
07. epd2in13_V2  True
08. epd2in13_V3  True
09. epd2in13b_V3 True
10. epd2in13bc   True
11. epd2in13d    True
12. epd2in66     True 
....
```

txoof, 2022-02-15:
Or maybe wait on this until we roll in Omni EPD.

aaronr8684, 2022-02-15:
I like this for now. The Omni module has a similar function that we could replace this with when we switch over. What do you think about adding a verbose flag to the `list_compatible_modules` function that defaults to True, but could be set to False if we just want a list of support without the what/why?

aaronr8684, 2022-02-16:
I'm not sure the second item is really needed. See my note in #9

closed 2022-02-16 by aaronr8684