txoof/PaperPi #25: Add the option for daemon (default) or user run

aaronr8684, 2022-02-17:
I think it would be a nice to allow the user to choose if they don't want to install this as a service automatically, but give them the necessary commands to install it later if they change their mind.

txoof, 2022-02-17:
I think after we sort this we'll be ready to squash the manage_modules branch and merge it with the main branch.  What do you think?
  
The plugin installer isn't super important at the moment as there's no big back-log of plugin submissions so I can manually merge in a few for now.

Then it's on to Omni EPD!

aaronr8684, 2022-02-17:
Yea, let me work on this now. I just ran a quick test of the install.sh and it seems like everything is working as expected still.

txoof, 2022-02-17:
resolved with PR #27

PR #27 (2022-02-17): I#25 prompt for daemon
Resolves #25
closed 2022-02-17 by txoof