txoof/PaperPi #35: Conflicting 'paperpi.ini' configs causes a weird behavior

aaronr8684, 2022-03-16:
I ran into a weird bug where if two config files have different screens, the first few operations will run on the wrong driver/screen config and then by the time the first plugin is displayed, it works as intended. I was able to replicate the issue (after fixing it). It's also likely not a high impact bug since I'm sure the percentage of people running two different screens on the same device is small. Anyway, here are the details and steps to reproduce...

Issue found: The first "clear screen" operation in the Screen class does not complete successfully when two different screen types are listed in two separate paperpi.ini files. I was using a 4.2" yellow screen on a 32-bit Pi4 running buster. It's not a fresh install and there might be some issues with how I'm running the PaperPi application, but I wanted to at least document it.

Console output:
``` bash
pi@raspberrypi:~/github/PaperPi $ python paperpi/paperpi.py
18:51:40 paperpi :get_config_files:213 :DEBUG      - gathering configuration files
18:51:40 paperpi :get_config_files:249 :INFO       - using configuration files to configure PaperPi: [PosixPath('/home/pi/github/PaperPi/paperpi/config/paperpi.ini'), PosixPath('/home/pi/.config/com.txoof.paperpi/paperpi.ini')]
18:51:40 paperpi :main      :622 :INFO       - ********** PaperPi 0.3.0.0 Starting **********
18:51:40 paperpi :main      :623 :DEBUG      - configuration:
{'main': {'display_type': 'waveshare_epd.epd4in2c', 'vcom': 0.0, 'max_refresh': 4, 'log_level': 'DEBUG', 'splash': False, 'rotation': 0, 'daemon': False}, 'Plugin: default fallback plugin': {'layout': 'layout', 'plugin': 'default', 'refresh_rate': 30, 'min_display_time': 60, 'max_priority': 32768}, 'Plugin: Reddit Quotes': {'layout': 'layout', 'plugin': 'reddit_quote', 'refresh_rate': 100, 'min_display_time': 50, 'max_priority': 2, 'max_length': 144}, 'xPlugin: Librespot': {'layout': 'layout', 'plugin': 'librespot_client', 'refresh_rate': 10, 'max_priority': 0, 'min_display_time': 15, 'player_name': 'SpoCon-Spotify', 'idle_timeout': 10}, 'Plugin: Moon Phase': {'layout': 'layout', 'plugin': 'moon_phase', 'min_display_time': 30, 'max_priority': 2, 'email': 'you@host.diamond', 'location_name': 'Europe/Amsterdam', 'lat': 52.3, 'lon': 4.9}, 'xPlugin: LMS - Your Player Name': {'layout': 'layout', 'plugin': 'lms_client', 'player_name': 'Your Player Name', 'refresh_rate': 5, 'min_display_time': 30, 'max_priority': 0, 'idle_timeout': 15}, 'xPlugin: Basic Clock': {'layout': 'layout', 'plugin': 'basic_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'Plugin: XKCD Comic Plugin': {'layout': 'layout', 'plugin': 'xkcd_comic', 'refresh_rate': 1200, 'min_display_time': 120, 'max_x': 800, 'max_y': 600, 'resize': 0, 'max_retries': 10, 'max_priority': 2}, 'xPlugin: Word Clock': {'layout': 'layout', 'plugin': 'word_clock', 'refresh_rate': 125, 'min_display_time': 255, 'max_priority': 2}, 'xPlugin: A Demo Plugin': {'layout': 'layout', 'plugin': 'demo_plugin', 'min_display_time': 30, 'max_priority': 1, 'your_name': 'Slartybartfast', 'your_color': 'chartreuse'}, 'xPlugin: New Yorker Comic': {'layout': 'layout', 'plugin': 'newyorker', 'day_range': 5, 'refresh_rate': 120, 'min_display_time': 60, 'max_priority': 2}, 'xPlugin: Weather Adis Ababa': {'layout': 'layout', 'plugin': 'met_no', 'refresh_rate': 300, 'min_display_time': 40, 'max_priority': 2, 'location_name': 'Adis Ababa', 'lat': 9.0, 'lon': 38.75, 'email': 'you@host.diamond'}, 'Plugin: Crypto Bitcoin v USD': {'plugin': 'crypto', 'layout': 'layout', 'fiat': 'usd', 'coin': 'bitcoin', 'days': 14, 'interval': 'hourly', 'rss_feed': 'https://bitcoinmagazine.com/.rss/full/', 'min_display_time': 30, 'refresh_rate': 300, 'max_priority': 2}, 'xPlugin: decimal binary clock': {'layout': 'layout', 'plugin': 'dec_bin_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, '__cmd_line': {'list_plugins': False}}

18:51:40 Screen  :_loadEPD  :373 :DEBUG      - configuring omni_edp.waveshare_epd.epd4in2c
18:51:40 Screen  :epd       :312 :DEBUG      - epd configuration waveshare_epd.epd4in2bc
18:51:40 Screen  :rotation  :359 :DEBUG      - rotation=0, resolution=[400, 300]
18:51:40 Screen  :wrapper   :247 :DEBUG      - initing display
18:51:40 epd4in2bc:ReadBusy  :70  :DEBUG      - e-Paper busy
18:51:40 epd4in2bc:ReadBusy  :73  :DEBUG      - e-Paper busy release
18:51:40 Screen  :clearEPD  :398 :DEBUG      - clearing screen
18:51:41 epd4in2bc:ReadBusy  :70  :DEBUG      - e-Paper busy
``` 

First few lines of `/etc/default/paperpi.ini`:
``` ini
# CONFIG_VERSION=1
[main]
# waveshare display type use HD for IT8951 displays
display_type = waveshare_epd.epd7in5b_V2

```

First few lines of `/home/pi/.config/com.txoof.paperpi/paperpi.ini`:
``` ini
# CONFIG_VERSION=1
[main]
# waveshare display type use HD for IT8951 displays
display_type = waveshare_epd.epd4in2c
```

Changing the `/etc/default/paperpi.ini` to the `waveshare_epd.epd4in2c` driver fixes the issue.

aaronr8684, 2022-03-16:
It's also a little weird that even after changing the config in /etc/default and then rerunning the program, the first clear still doesn't behave correctly until the first plugin is displayed. BUT if I stop the program and run it again (changing nothing), it works as intended.

txoof, 2022-03-17:
This is indeed super weird. The /etc version and ~/ version should be mutually exclusive. The logic **should** be to load the appropriate version based on daemon or user mode.

I will investigate further.

txoof, 2022-03-17:
Can you check the first few lines of the debug output? It should list the config files that are loaded:

This is my output when I run it in user-land mode.
```TEXT
20:13:30 paperpi :get_config_files:249 :INFO       - using configuration files to configure PaperPi: [PosixPath('/home/pi/src/paperpi/paperpi/config/paperpi.ini'), PosixPath('/home/pi/.config/com.txoof.paperpi/paperpi.ini')]
```
I'm curious to see what shows up there on your system.

aaronr8684, 2022-03-17:
Same...it's in the original report, you just have to scroll over. I was messing with the logging.cfg file and is wasn't there in the original post (and probably email notification), but I edited it a few mins later with the first few lines added.

aaronr8684, 2022-03-17:
Honestly, I wouldn't bother chasing this down too much. It could be just a weird issue in a very specific situation. When I do a full install test with EPDLib v0.6, I'll see if I can replicate the issue

closed 2026-11-09 by txoof-bot