txoof/PaperPi #37: Confirm greyscale / color support in EPDLib doesn't break PaperPi

aaronr8684, 2022-03-17:
Confirm grey and color support
- [ ] Pending EPDLib publishing to PyPi
- [ ] Test install process (32/64)
- [ ] Confirm multiple screens

Screens/modes tested:
- [ ] 2.13 BW
- [ ] 3.7 BW
- [ ] 4.2 BW
- [ ] 5.65 BW
- [ ] 7.5 BW

txoof, 2022-03-17:
@aaronr8684 modified epdlib works, but breaks grayscale support in paperpi for IT8951 

Also need to update the documentation for the format for `display_type` in the configuration.

txoof, 2022-03-17:
Also, splash screen doesn't appear and nothing displays until the first plugin is loaded. I suspect this is related to the bug you noticed in #35. This error cropped up immediately after moving to epdlib 0.6.

aaronr8684, 2022-03-17:
Ok. I figured greyscale was likely broken because of the combined logic, but will be fixed in the next update with color support.

aaronr8684, 2022-03-17:
> Also, splash screen doesn't appear and nothing displays until the first plugin is loaded. I suspect this is related to the bug you noticed in #35. This error cropped up immediately after moving to epdlib 0.6.

Splash works fine for me on the 7.5b screen, let me try on the 4.2 with the correct config.

txoof, 2022-03-17:
looking forward to that. I think the logic in paperpi will probably need some help too for grayscale. 

I think the problem is located in [paperpi.py](https://github.com/txoof/PaperPi/blob/b48c263607fdba81bc29e406eb7f37b2c9e33b1b/paperpi/paperpi.py#L336-L338). If "HD" isn't in `display_type`, it passes the `force_onebit` param to the Plugin object.

```
            # force layout to one-bit mode for non-HD screens
            if not config['main'].get('display_type') == 'HD':
                plugin_config['force_onebit'] = True
```

We could just fudge it and set `force_onebit=False` for all cases and see what happens :)

aaronr8684, 2022-03-17:
> looking forward to that. I think the logic in paperpi will probably need some help too for grayscale.

Yea absolutely (and for color). I'm still thinking how we could allow for color support but also default to BW if nothing is specified. I'm thinking the default logic will stay in EPDLib (for back support) and the various modes can be specified in PaperPi if needed.

> We could just fudge it and set `force_onebit=False` for all cases and see what happens :)

or just change the logic to `if 'it8951' not in config['main'].get('display_type'):`

That's the same logic I used (but without the 'not') in EPDLib for the time being to prompt for a VCOM

aaronr8684, 2022-03-17:
I also made one change to the image mode in the EPDLib that might help re-enable greyscale.

aaronr8684, 2022-03-17:
4.2c splash works as expected with the correct config

txoof, 2023-06-19:
I've merged in RGB support without OMNI EPD #79

closed 2023-06-19 by txoof