txoof/PaperPi #43: -c/--config option does not work properly

txoof, 2022-03-27:
cannot specify alternative config file at command line always defaults to either daemon or user config.

aaronr8684, 2022-03-27:
I noticed this as well. When I noticed it, I started going through the config logic to see if we could use the daemon config as a fallback in case the user doesn't edit the config as prompted for #42

txoof, 2022-05-02:
resolved by [5793147](https://github.com/txoof/PaperPi/commit/5793147adfee4e541db48b2e925bbcba6547ff83)

txoof, 2022-05-02:
Not sure if falling back is a good idea; the daemon version will likely also be misconfigured if the user didn't bother to edit the config as prompted.

closed 2022-05-02 by txoof

aaronr8684, 2022-05-02:
Were you able to resolve the other issue where it appeared to grab config options from 2 different files? It seemed to resolve after letting the init run at least once

txoof, 2022-05-02:
I haven't looked at that one yet. I'll take a look at it tomorrow. I suspect this might fix the problem, but I'll need to dig into the issue some more.