txoof/PaperPi #51: disable Full update for epd2in13_V2

adi170-alt, 2022-05-04:
It annoys me that my screen has to fully update whenever something has to change on the screen, I don't think it is that bad, but could I try it? it will maybe be awful, but I would like to try it, maybe it isn't that bad. I could try to change it on my own but that could take me hours, and you guys have more understanding of the code.

txoof, 2022-05-04:
@adi170-alt Hmm. That's indeed possible, but would need to be managed in the backend [epdlib](https://github.com/txoof/epdlib). 

I'd be interested in making PaperPi work with this, but I don't have a 2in13_V2 to test on. I can support you in getting started on hacking this. epdlib is pretty well documented and the changes you would need to make are in the [Screen class ](https://github.com/txoof/epdlib/blob/843807aa7b09272f2bf9c18750f70dcae2d839f3/epdlib/Screen.py#L192). 

When we switch to OmniEPD, this might be more realistic. Right now OmniEPD doesn't support grayscale and partial updates.

adi170-alt, 2022-05-04:
> 

thank u for the info,
how can I change my fork into something like this for example?: import something

txoof, 2022-05-05:
If you want to use PaperPi against your own fork of epdlib, build out the PaperPi environment using the scripts in the utilities directory, then from the PaperPi directory run:
```
pipenv uninstall epdlib --skip-lock
pipenv install -e git+https://path/to.your.git#egg=epdlib --skip-lock
```

I've found that pipenv hangs on the locking when PILLOW is in the dependencies; for the purposes of this project a .lock file is not needed hence the `--skip-lock`

closed 2023-03-13 by txoof