txoof/PaperPi #58: Show IP address

ZubaZ21, 2022-11-19:
Either on the demo plugin or on the splash page, show the IP of the PaperPi for easy connection.

txoof, 2023-06-20:
See #93

txoof, 2023-06-22:
@ZubaZ21 I created a plugin that shows system information. 

I thought about rolling this into the splash screen, but didn't love this idea for the following reasons:

- Splash screen only shows for less than a minute and only when the application starts, limiting the usefulness of adding information like ip address; it's easy to miss
- The splash screen has very few dependencies and really, really simple. It's only about 30 lines of code and most of that is a docstring and white space
- Anyone willing to install a daughter board and an application from the command line should be able to find the ip address of their device. 
- A plugin that displays this has more functionality and can regularly be in the rotation; though I suppose if you **really** wanted to, you could jam the splash screen into the rotation. It would work.

If you have any other feels about this, let me know. I'm open to reconsidering.

closed 2023-06-22 by txoof