txoof/PaperPi #70: Screen Mirrored issue

mTyguy, 2022-12-05:


txoof, 2022-12-05:
Good news: this can be fixed by adding a mirror option to the [epdlib.Screen class](https://github.com/txoof/epdlib/blob/843807aa7b09272f2bf9c18750f70dcae2d839f3/epdlib/Screen.py#L541-L544
)

I'll be traveling for work this week, so it might not be until the weekend/next week until I can get to this.

txoof, 2022-12-05:
Aaand I'm disorganized. Turns out there's already a solution for this; it just needs to be tested and merged in to epdlib and then integrated into the main branch here.

mTyguy, 2022-12-06:
that is great news. everything else is working great!

txoof, 2022-12-12:
Good news!

I'll do some more tests at my end and then roll the changes into the main branch.

txoof, 2022-12-12:
resolved with c6d5e360f571699c0d35cf08f7cef90d9878edcf