txoof/PaperPi #71: epd2in9d text is way too tiny

mattyteaks, 2022-12-07:
Understanding this is a small screen, I can not figure out how to get the font size up to a readable size on even the most basic plugins.

For example, the librespot plugin, in just trying to use the two_rows_text_only layout of the librespot_client plugin, the song title is perfectly readable, but the artist name is in a tiny unreadable size, all the way at the bottom.  The artist name is more legible in the default layout with album art than it is here, but I can not figure out how to properly get the fonts to render.

Understanding this might just be my novice ignorance, any ideas or maybe a nudge in the right direction?

txoof, 2022-12-08:
@mattyteaks 
There isn't a way to directly control the font size for a given layout. The font size is **calculated** based on the width and height of the font and number of pixels available to the layout block. That is to say, the best way to influence the font size in a block is to:
1. Increase the amount of space available for a font
2. Choose a font that has better proportions and letter sizing

You can try playing with the two_rows_text_only layout in `/usr/local/paperpi/plugins/librespot_client/layout.py`. Try adjusting line [12](https://github.com/txoof/PaperPi/blob/8a6a905d1a640c20a82a1c91deeb24e4ccadfcac/paperpi/plugins/librespot_client/layout.py#L12) and line [27](https://github.com/txoof/PaperPi/blob/8a6a905d1a640c20a82a1c91deeb24e4ccadfcac/paperpi/plugins/librespot_client/layout.py#L27). They should add up to exactly 1.0.

You can also try removing the padding (set it to 0). Finally, try a more slender font. [Economica](https://fonts.google.com/specimen/Economica?query=econo) might fit better in that space.

The easiest way to test without setting up a development environment is to make a change, then restart paperpi: `sudo systemctl restart paperpi-daemon.service`

You can find documentation for working with layouts [here](https://github.com/txoof/epdlib#quick-start-recipes)

If you have something that works better on your screen, I would be really appreciate it if you shared it back with the project. I'll make to roll it into the layout.

mattyteaks, 2022-12-08:
Alright, awesome! Thanks for that. That's definitely got me on the right path.
Turns out, I was not working with the layouts under the root user.

Will play around with various settings and definitely share what I can get to work best for the text-driven plugins and splash screen on smaller screens.

txoof, 2022-12-09:
You can also try fiddling with the max_lines parameter. It's currently see to 2 lines, but maybe 1 will work for you.

txoof, 2022-12-11:
@mattyteaks I'm closing this issue, but if you have any suggestions for improving the layout, feel free to add them as a PR, or a comment here.

closed 2022-12-11 by txoof