txoof/PaperPi #72: Can slideshow pull images from pendrive?

PaperCloud10, 2022-12-10:
Hello,

It seems that when specifying a mounted drive as a directory for the slideshow plugin permission is denied. When specifying the default directory it works fine. This is my first raspberry pi project and I'm not sure how to proceed from here, is it possible to enable permissions for the slideshow plugin to access images from mounted drives?

Dec 10 18:37:18 raspberrypi systemd[1]: Started PaperPi E-Paper Display.
Dec 10 18:37:25 raspberrypi paperpi[1770]: 18:37:25 paperpi :get_config_files:251 :INFO       - using configuration files to configure PaperPi: [PosixPath('/usr/local/paperpi/config/paperpi.ini'), PosixPath('/etc/default/paperpi.ini')]
Dec 10 18:37:43 raspberrypi paperpi[1770]: Traceback (most recent call last):
Dec 10 18:37:43 raspberrypi paperpi[1770]:   File "/usr/local/paperpi/paperpi.py", line 759, in <module>
Dec 10 18:37:43 raspberrypi paperpi[1770]:     exit_code = main()
Dec 10 18:37:43 raspberrypi paperpi[1770]:   File "/usr/local/paperpi/paperpi.py", line 732, in main
Dec 10 18:37:43 raspberrypi paperpi[1770]:     plugins = build_plugins_list(config=config, resolution=screen.resolution,
Dec 10 18:37:43 raspberrypi paperpi[1770]:   File "/usr/local/paperpi/paperpi.py", line 431, in build_plugins_list
Dec 10 18:37:43 raspberrypi paperpi[1770]:     my_plugin.update()
Dec 10 18:37:43 raspberrypi paperpi[1770]:   File "/usr/local/paperpi/library/Plugin.py", line 165, in update
Dec 10 18:37:43 raspberrypi paperpi[1770]:     is_updated, data, priority = self.update_function(*args, **kwargs)
Dec 10 18:37:43 raspberrypi paperpi[1770]:   File "/usr/local/paperpi/plugins/slideshow/slideshow.py", line 265, in update_function
Dec 10 18:37:43 raspberrypi paperpi[1770]:     image_array = _index_images(image_path)
Dec 10 18:37:43 raspberrypi paperpi[1770]:   File "/usr/local/paperpi/plugins/slideshow/slideshow.py", line 44, in _index_images
Dec 10 18:37:43 raspberrypi paperpi[1770]:     if image_path.is_dir():
Dec 10 18:37:43 raspberrypi paperpi[1770]:   File "/usr/lib/python3.9/pathlib.py", line 1422, in is_dir
Dec 10 18:37:43 raspberrypi paperpi[1770]:     return S_ISDIR(self.stat().st_mode)
Dec 10 18:37:43 raspberrypi paperpi[1770]:   File "/usr/lib/python3.9/pathlib.py", line 1221, in stat
Dec 10 18:37:43 raspberrypi paperpi[1770]:     return self._accessor.stat(self)
Dec 10 18:37:43 raspberrypi paperpi[1770]: PermissionError: [Errno 13] Permission denied: '/media/pi/Pendrive'
Dec 10 18:37:44 raspberrypi systemd[1]: paperpi-daemon.service: Main process exited, code=exited, status=1/FAILURE
Dec 10 18:37:44 raspberrypi systemd[1]: paperpi-daemon.service: Failed with result 'exit-code'.
Dec 10 18:37:44 raspberrypi systemd[1]: paperpi-daemon.service: Consumed 20.450s CPU time

txoof, 2022-12-10:
@PaperCloud10 
Nice catch on that error. It most definitely should not crash when you get a `Permission denied` error. I'll start work on fixing that and add some documentation around this.

It looks like you're running in daemon mode. Can you confirm that?

To answer your question simply: yes, PaperPi can pull images from anywhere. It doesn't care one bit where the files are located. The solution to your  problem is in the error message `PermissionError: [Errno 13] Permission denied: '/media/pi/Pendrive'` The PaperPi daemon does not have read access to the path that you have specified. You have two options to fix this:

1. Change the permissions on the directory and all of its contents so anyone can read it - since this is a mounted volume
2. Make sure that the PaperPi user is a member of the group that owns `/media/pi/Pendrive`

## changing permissions

If there's anything "secret" in this directory, you'll

txoof, 2022-12-10:
additional unhandled exception:
```text
Traceback (most recent call last):
  File "/usr/local/paperpi/paperpi.py", line 759, in <module>
    exit_code = main()
  File "/usr/local/paperpi/paperpi.py", line 735, in main
    exit_code = update_loop(plugins=plugins, screen=screen, max_refresh=config['main'].get('max_refresh', 5))
  File "/usr/local/paperpi/paperpi.py", line 571, in update_loop
    priority_list = _update_plugins()
  File "/usr/local/paperpi/paperpi.py", line 530, in _update_plugins
    plugin.update()
  File "/usr/local/paperpi/library/Plugin.py", line 165, in update
    is_updated, data, priority = self.update_function(*args, **kwargs)
  File "/usr/local/paperpi/plugins/slideshow/slideshow.py", line 336, in update_function
    image = _add_border(current_image, frame)
  File "/usr/local/paperpi/plugins/slideshow/slideshow.py", line 103, in _add_border
    im = Image.open(image)
  File "/usr/local/paperpi/.venv/lib/python3.9/site-packages/PIL/Image.py", line 3186, in open
    raise UnidentifiedImageError(
PIL.UnidentifiedImageError: cannot identify image file '/media/drive/._140011-untitled shoot_5x7, Annemieke, Patti, Voorlinden, Wassenaar.jpg'
```

txoof, 2022-12-10:
@PaperCloud10 
How are you mounting the pen drive? Are you using the GUI window interface, or actually issuing `sudo mount /dev/sda1 /media/pi/Pendrive` or something similar?

txoof, 2022-12-10:
@PaperCloud10 
OK. I've been on an odyssey of learning the RaspberryPi GUI, but I have a solution if you're using the Graphical Interface to mount your drive:

# 1 - Disable PCManFM Auto Mounting

**WARNING** These steps will make it so any user on the Pi can read the USB drive. If you're OK with this, proceed.

PCManFM, the default file manager, will only mount external USB devices with permissions that make it impossible for other users to read your USB drive. 

1. eject any USB sticks/external drives you have attached
2. Open PCManFM by double clicking on the icon or running `pcmanfm` from a terminal window
3. Click Edit > Preferences > Volume Management
4. Uncheck "Mount removable media automatically when they are inserted" and "Show available options for removable media when they are inserted"
5. Install usbmount `sudo apt install usbmount`
6. confirm the permissions on `/media/pi' by running `ls -alh /media/` You should see output like below. The important thing is that the permissions are set as `drwxr-x-rx` for the `pi` directory. If they are not run `sudo chmod 755 /media/pi` to fix them.
```text
$ ls -alh /media
drwxr-xr-x  3 root root 4.0K Dec 10 12:42 .
drwxr-xr-x 18 root root 4.0K Sep 22 02:25 ..
drwxr-xr-x  3 root root 8.0K Jan  1  1970 pi
```
7. Insert a USB drive. The default should be to mount with the permissions rwxr-xr-x which should allow all users to use the removable media.

Let me know if that helps. I'll work on fixing the exceptions you caught.

txoof, 2022-12-10:
Resolved with #ec1aa52663894106873f85c9dedf7a50d8075583

txoof, 2022-12-11:
See the documentation here: https://github.com/txoof/PaperPi/blob/main/paperpi/plugins/slideshow/README.md#removable-media

closed 2022-12-11 by txoof

PaperCloud10, 2022-12-11:
@txoof I have just tested this today and it works! Thank you kindly, your help is much appreciated.

txoof, 2022-12-11:
Amazing how much the GUI can do for you, and how much it can get in they way in unexpected ways!

I live mostly on the command line, so this was all new for me. Learned a whole lot!