txoof/PaperPi #76: Documentation error.

blbal, 2023-01-03:
In a number of places across documentation for this project (Plugins.md, step_by_step_instructions.md) the location of the configuration file is (incorrectly) referred to as /etc/defaults/paperpi.ini instead of /etc/default/paperpi.ini. Presume this is a typo?

txoof, 2023-01-04:
@blbal thanks for catching this. It should all be updated in 7b2e925.

Did you manage to get it installed?

closed 2023-01-04 by txoof

blbal, 2023-01-04:
Thanks, yes, I did get it installed - helps that I have used and abused Debian for more years than I care to remember :wink:  Thanks for the excellent code. Just wish I was better acquainted with python, but I did manage to create a slightly modified `basic clock` on a pizero2W/epd_2in13b_V3, which was the ultimate aim.

txoof, 2023-01-05:
@blbal Glad you managed to get it setup and thanks for the compliments.

What did you modify in the Basic Clock? I'm keen to add more plugins or modify the plugins so they are more useful. Perhaps I can roll some of your modifications into the project.

If you're interested in fiddling a bit of Python, developing a plugin might be a good place to start.