txoof/PaperPi #83: LMS - WARNING    - error making connection to server: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'

ZubaZ21, 2023-03-04:
I have two Squeezebox players.  One is a original Boom and one is a PiCorePlayer.  While I could swear that the plugin worked at one point, I am not able to display the plugin now.  The devices are all on the same network (192.168.3.x). Weather, Word Clock, and Reddit quotes are all working as designed.

My plugin config (I have tried the same config with the Boom's name listed with no change.):

> [Plugin: LMS - ZuCorePlayer]
> layout = layout
> plugin = lms_client
> player_name = ZuCorePlayer
> refresh_rate = 5
> min_display_time = 10
> max_priority = 0
> idle_timeout = 15

daemon.log:

> Mar  4 12:07:02 PaperPi paperpi[12751]: 12:07:02 paperpi :_update_plugins:451 :INFO       - ##########Plugin: LMS - ZuCorePlayer##########
> Mar  4 12:07:02 PaperPi paperpi[12751]: 12:07:02 lms_client:update_function:110 :DEBUG      - update_function for plugin Plugin: LMS - ZuCorePlayer, version 0.1.2
> Mar  4 12:07:02 PaperPi paperpi[12751]: 12:07:02 lms_client:update_function:143 :DEBUG      - LMS Service created
> Mar  4 12:07:03 PaperPi paperpi[12751]: 12:07:03 QueryLMS:query     :270 :WARNING    - error making connection to server: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
> Mar  4 12:07:03 PaperPi paperpi[12751]: 12:07:03 lms_client:update_function:193 :DEBUG      - now_playing: None
> Mar  4 12:07:03 PaperPi paperpi[12751]: 12:07:03 lms_client:update_function:222 :INFO       - priority set to: 32768
> Mar  4 12:07:03 PaperPi paperpi[12751]: 12:07:03 paperpi :_update_plugins:458 :INFO       -      PRIORITY: 32768 of max 0
> Mar  4 12:07:03 PaperPi paperpi[12751]: 12:07:03 paperpi :_update_plugins:461 :DEBUG      -      DATA: 11 elements
> Mar  4 12:07:03 PaperPi paperpi[12751]: 12:07:03 paperpi :_update_plugins:462 :DEBUG      -      IMAGE: <PIL.Image.Image image mode=L size=800x480 at 0xB5473118>

Any thoughts?

ZubaZ21, 2023-03-04:
A bit more info . . . 
I have 
1. restarted the paperpi service
2. Rebooted both squeeze devices
3. I am able to control both devices via http get calls

ZubaZ21, 2023-03-04:
Re-ran the installer (last install was long, long ago) - Rebooted everything.
Things are working.
User error?

ZubaZ21, 2023-03-04:
So . . . 
Not resolved.
The PaperPi plugin is showing now playing info for local music and Spotify (through the Spotty plugin) but not some streaming music.
Tested with Pandora and streaming radio (KXT (http://kera-ice.streamguys.us/kxtlive128.m3u) and KUZU (http://opml.radiotime.com/Tune.ashx?id=s295557&formats=ogg,mp3,wma,wmvoice&partnerId=16&serial=b88aed1220c19f3d4751f40665d4b568))
The "now playing" info shows on all the devices for Pandora and the radio streams . . . . but not for the Plugin

txoof, 2023-03-05:
@ZubaZ21 

Let me make sure I understand the issue:
- PaperPi is generally working
- librespot client is displaying spotify content as expected
- lms_client is displaying local streams (e.g. MP3s played from the Logitech Media Player on the Boom `ZuCorePlayer`
- lms_client is **NOT** displaying Pandora or Radio streams

I believe the problem related to the streams probably comes from some subtle difference with how the LMS player delivers the stream data. I definitely never tested the QueryLMS module against streams. 

Do you know if the pandora/radio streams ever worked in the past? 

I don't have a Pandora account; it's not currently available in the Netherlands so I can't directly check that. I'll have to have a play with the KUZU stream and see if I can sort out what the difference is.

Also KUZU has a pretty great stream going on right now (20:30 GMT) on 5 March. Thanks for the tip! I just added it to my favorites. I forgot how good a good DJ can be.

ZubaZ21, 2023-03-05:
- Paperpi is generally working
- librespot is enabled, but I listen to Spotify througn LMS via a plugin on that server (https://github.com/michaelherger/Spotty-Plugin)
- lms_client displays local mp3 data
- lms_client displays Spotify data (probably through the above plugin)
- lms_client does not display Pandora content.  I am unaware if it ever did; I do not recall ever trying to display it on paperpi.  I do not see the plugin during the rotation.  Pandora now-playing information shows on the two players I have.
- lms_client does not show data for the two radio streams I have tried; I do not recall ever trying to display it on paperpi.  I do not see the plugin during the rotation. (the other test stream - https://kera.streamguys1.com/kxtlive128 ). Pandora now-playing information shows on the two players I have.

ZubaZ21, 2023-03-05:
[DS82_Complete.pdf](https://github.com/txoof/PaperPi/files/10893107/DS82_Complete.pdf)
Attaching /etc/default/PaperPi.ini (running as daemon)

txoof, 2023-03-06:
@ZubaZ21 I think you may have attached the wrong content in the previous comment. I deleted the comment because it looked like some personal documents. 

Thanks for the update. I'm rather surprised that spotify shows up through the lms_client. I'll need to explore that further.  I have an idea what's going on with pandora and the radio streams, but it will take more investigation to get to the bottom of it.

I wouldn't mind seeing your paperpi.ini either.

ZubaZ21, 2023-03-06:
[paperpi.ini.txt](https://github.com/txoof/PaperPi/files/10897022/paperpi.ini.txt)
Thanks for cleaning up my mess.

Here is the ini file.

txoof, 2023-03-06:
After looking at the QueryLMS code which I borrowed heavily from a defunct project, I see that there's a bunch of [undocumented methods](https://github.com/txoof/querylms/blob/b3c0bb587d76bf71cccf647292a4e286f0e0f7d5/QueryLMS/QueryLMS.py#L594-L621
) for querying the radio status. I suspect this is where things go wrong. 

When I wrap up some other projects, I'll dig into this one and see if I can't determine exactly the problem and then look into adding this functionality to the lms_client plugin.

Feel free to dive in if you're motivated!

ZubaZ21, 2023-03-06:
I will look into things but I am more of a hack-at-things than know-what-I'm-doing
Another open source app I use that displays LMS content is https://github.com/kaaholst/android-squeezer
It's maintained and shows the data that is missing from the plug-in

txoof, 2023-03-06:
The place to start would be the LMS client plugin.

You can clone the repo onto a pi and run the `utilities/create_devel_environment.sh` script if you use `create_devel_environment.sh -j` it will install and setup a jupyter kernel and allow you to hack on the ipynb files.

I'm working on an update to PaperPi that supports color output right now and focused on getting it ready to publish. Once I get that under control, I'll loop back around to this.

txoof, 2023-03-13:
I've refactored the `querylms` package to handle streams now. It should be a drop-in update for paperpi.

I'd suggest installing from the development branch and testing that. 

If you want to try to upgrade in place, you'll need to do the following:

```bash
$ sudo su -
$ cd /usr/local/paperpi 
$ pipenv update
```

It's possible that pipenv will fail miserably while trying to generate a lock file; in that case, you can fight with pipenv or just do a fresh install from the dev branch: 

```
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/txoof/PaperPi/development/install/remote_install_development.sh)"
```

ZubaZ21, 2023-03-13:
This is resolved.
Tested with local server content, streaming radio, Spotify, and Pandora

closed 2023-03-13 by ZubaZ21

txoof, 2023-03-14:
@ZubaZ21 Great to hear!

I did notice that the album title sometimes gets stuck and I'm not entirely sure why. It was a pretty easy fix once I reacquainted myself with the LMS API.