txoof/PaperPi #86: epd7in3f support？

cloud263, 2023-03-29:
May I ask whether it supports epd7in3f now, which is Waveshare's new ink screen.

txoof, 2023-03-30:
@cloud263 Yes! The development branch supports all the f screens. I need a few more people to test it out and then I can merge the changes into the production version. If you have some time to test it out and give some feedback, I would greatly appreciate it!

You can join the discussion here: https://github.com/txoof/PaperPi/discussions/77

And you can install the development branch using this curl installer: 

```
 /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/txoof/PaperPi/I%2379_color_display/install/remote_install_development.sh)"
```

txoof, 2023-03-31:
@cloud263 Any luck trying the development version?