txoof/PaperPi #87: Shutdown without cleaning screen

TychoWerner, 2023-03-29:
Hi,

I am trying to use the screen as photo display. Only requiring a change every month or so.
When I shutdown the system PaperPi starts a clean_up and the screen is wiped.
Is there a way to shutdown PaperPi and let the last screen remain?

Hope to hear from you

txoof, 2023-03-30:
@TychoWerner 

That's not currently possible, but would be pretty trivial to add. 

It would just take a modification of the [`cleanup()` function](https://github.com/txoof/PaperPi/blob/a695fe48ca90dfa54b84253c262f6e7977e3e4ee/paperpi/paperpi.py#L337-L358
) and a command line switch and config file param. 

PRs are welcome; would you like to take a stab at it?

txoof, 2023-03-30:
That would work, but a more elegant approach would be to just comment out the clear screen command in clean_up. Then the cache is properly handled and you don't end up filling up your storage with garbage.

Thought the temp files **should** be deleted on reboot.

txoof, 2023-06-23:
@TychoWerner I've merged an update for this into the development branch. The update does the cleanup and clears the screen by default. This can adjusted in the configuration `.ini` file.

after I test this a bit more, I'll merge it into the main branch.

txoof, 2023-07-25:
added in b6d5c021e5a4a8f3b13db789a594ba862e253e88

closed 2023-07-25 by txoof