txoof/PaperPi #89: Unable to get anything on e-paper display after paperpi installation (Display: epd2in13_V2)

aksakure, 2023-06-07:
$ paperpi -d
01:04:49 paperpi :get_config_files:255 :INFO       - using configuration files to configure PaperPi: [PosixPath('/usr/local/paperpi/config/paperpi.ini'), PosixPath('/etc/default/paperpi.ini')]
01:04:52 Block   :padding   :346 :WARNING    - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
01:04:52 Block   :padding   :353 :WARNING    - the padded area available may be too small to display any content: Area: (250, 12), Padded Area: [230, -8]
01:04:52 Block   :font_size :824 :WARNING    - no font size set, using {font_size}

txoof, 2023-06-08:
Looks like you're trying to display a high resolution/data-rich plugin on a small screen, but it's hard to tell with the information you've provided. 

Can you share some more information? In particular:
* Screen model
* The `[main]` section of your configuration file

You can paste your config file directly into the comment, but 3x "back tick" [  ` ] characters before and after it to help make it easier to read. It will render like this:
```
$ paperpi -d
01:04:49 paperpi :get_config_files:255 :INFO       - using configuration files to configure PaperPi: [PosixPath('/usr/local/paperpi/config/paperpi.ini'), PosixPath('/etc/default/paperpi.ini')]
01:04:52 Block   :padding   :346 :WARNING    - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
01:04:52 Block   :padding   :353 :WARNING    - the padded area available may be too small to display any content: Area: (250, 12), Padded Area: [230, -8]
01:04:52 Block   :font_size :824 :WARNING    - no font size set, using {font_size}
```
instead of this mess:
$ paperpi -d
01:04:49 paperpi :get_config_files:255 :INFO       - using configuration files to configure PaperPi: [PosixPath('/usr/local/paperpi/config/paperpi.ini'), PosixPath('/etc/default/paperpi.ini')]
01:04:52 Block   :padding   :346 :WARNING    - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
01:04:52 Block   :padding   :353 :WARNING    - the padded area available may be too small to display any content: Area: (250, 12), Padded Area: [230, -8]
01:04:52 Block   :font_size :824 :WARNING    - no font size set, using {font_size}

txoof, 2023-06-08:
It would be good to know which plugins you are trying to use as well.

txoof, 2023-06-09:
Sorry, I missed it the first time: You're using a 2in13 model. Some of the default plugin layouts won't display well on that screen. 
Do you get anything at all at start up?

closed 2023-06-19 by txoof