txoof/PaperPi #92: Moon phases plugin broken

txoof, 2023-06-17:
Looks like MET.no api version has changed from V2.0 to V3.0; need to adjust endpoints and check for any differences in returned JSON.

jrmaxwell, 2023-06-17:
I had to make a small change to the plugin to get it working again today. The response code is coming back 203 instead of 200. Treating 203 as success seems to work but you'll probably want to check for differences in the API. This changed a few days ago but I just had chance to look into it today.

txoof, 2023-06-20:
The 200 -> 203 hack will work until sometime in August/September when the Sunrise V2 api is shutdown.

I've updated the plugin to the V3 API and pushed out a working version to the development branch. You can install it with this link `/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/txoof/PaperPi/main/install/remote_install_development.sh)"` 

Would you mind testing it before I push it to the main branch?

I really need to get around to separating the plugins from the main application and make a plugin installer/updater. That would make the upgrade less onerous.

jrmaxwell, 2023-06-21:
This reported the version as 0.5.1.0 and the moon phase plugin is working.

txoof, 2023-06-21:
Bumped the version number and merged development into main. Should be good to go.

Thanks for testing it.

closed 2023-06-21 by txoof

txoof, 2023-06-25:
@jrmaxwell 
I introduced a new bug that causes the moon display to kick an error when there is no moonset/rise in the following day. You may want to update again, or live with an error message some days.
[image]

from the [Sunrise API](https://docs.api.met.no/doc/sunrise/celestial.html)

## Lunar crossings
Due to the fact that the moon takes > 24 hours to complete a cycle on the sky, there are days (around the full moon) when the moon doesn’t cross the meridian, in case the high_moon values will return null`. Similar event will occur around the new moon (no crossing of antimeridian), as well as missing moonsets and moonrises.

txoof, 2023-06-25:
See #106