txoof/PaperPi #120: Remote Installation Script fails

VaporwareII, 2023-08-09:
**Describe the bug**
Install can't get past the PIP venv.  Tried twice but I don't understand the error messages presented around the PIP dependencies.

**To Reproduce**
Steps to reproduce the behavior:
1. Run Install Script
2. See error below

USER@Argon:~ $ /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/txoof/PaperPi/main/install/remote_install.sh)"
installing PaperPi from branch: main
cloning git repo into /tmp/PaperPi_rY5ya
Cloning into '/tmp/PaperPi_rY5ya'...
remote: Enumerating objects: 2020, done.
remote: Counting objects: 100% (495/495), done.
remote: Compressing objects: 100% (274/274), done.
remote: Total 2020 (delta 265), reused 399 (delta 218), pack-reused 1525
Receiving objects: 100% (2020/2020), 96.65 MiB | 4.11 MiB/s, done.
Resolving deltas: 100% (944/944), done.
Updating files: 100% (807/807), done.
sucessfully cloned branch https://github.com/txoof/PaperPi.git into /tmp/PaperPi_rY5ya
/tmp/PaperPi_rY5ya
[sudo] password for USER: 
==> /usr/bin/sudo /tmp/PaperPi_rY5ya/install/install.sh
checking if paperpi-daemon.service is running
   paperpi-daemon.service not running
done
checking for required debian packages
found debian packages for PaperPi module debian_packages-crypto.txt
checking libcairo2
found debian packages for PaperPi module debian_packages-paperpi.txt
checking libtiff5
checking libopenjp2-7
checking python3-pip
checking libatlas-base-dev
checking libjpeg-dev
required packages installed
installing required python packages with pip3
successfully installed requirements
Removing existing installation found at /usr/local/paperpi
Installing files to /usr/local/paperpi
Creating virtual environment for paperpi in /usr/local/paperpi
/usr/local/paperpi /home/USER
Creating a virtualenv for this project...
Pipfile: /usr/local/paperpi/Pipfile
Using /usr/bin/python3.9 (3.9.2) to create virtualenv...
⠼ Creating virtual environment...created virtual environment CPython3.9.2.final.0-64 in 596ms
  creator CPython3Posix(dest=/usr/local/paperpi/.venv, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/root/.local/share/virtualenv)
    added seed packages: pip==23.2.1, setuptools==68.0.0, wheel==0.41.0
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator

✔ Successfully created virtual environment!
Virtualenv location: /usr/local/paperpi/.venv
Pipfile.lock not found, creating...
Locking [packages] dependencies...
Building requirements...
Resolving dependencies...
Resolving dependencies...
✘ Locking Failed!
⠋ Locking...
CRITICAL:pipenv.patched.pip._internal.resolution.resolvelib.factory:Cannot install -r /tmp/pipenv-1f8cvnzx-requirements/pipenv-rdzy4lku-constraints.txt (line 3) because these package versions have conflicting dependencies.
[ResolutionFailure]:   File "/usr/local/lib/python3.9/dist-packages/pipenv/resolver.py", line 704, in _main
[ResolutionFailure]:       resolve_packages(
[ResolutionFailure]:   File "/usr/local/lib/python3.9/dist-packages/pipenv/resolver.py", line 662, in resolve_packages
[ResolutionFailure]:       results, resolver = resolve(
[ResolutionFailure]:   File "/usr/local/lib/python3.9/dist-packages/pipenv/resolver.py", line 642, in resolve
[ResolutionFailure]:       return resolve_deps(
[ResolutionFailure]:   File "/usr/local/lib/python3.9/dist-packages/pipenv/utils/resolver.py", line 1167, in resolve_deps
[ResolutionFailure]:       results, hashes, markers_lookup, resolver, skipped = actually_resolve_deps(
[ResolutionFailure]:   File "/usr/local/lib/python3.9/dist-packages/pipenv/utils/resolver.py", line 948, in actually_resolve_deps
[ResolutionFailure]:       resolver.resolve()
[ResolutionFailure]:   File "/usr/local/lib/python3.9/dist-packages/pipenv/utils/resolver.py", line 690, in resolve
[ResolutionFailure]:       raise ResolutionFailure(message=str(e))
[pipenv.exceptions.ResolutionFailure]: Warning: Your dependencies could not be resolved. You likely have a mismatch in your sub-dependencies.
  You can use $ pipenv run pip install <requirement_name> to bypass this mechanism, then run $ pipenv graph to inspect the versions actually installed in the virtualenv.
  Hint: try $ pipenv lock --pre if it is a pre-release dependency.
ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

Traceback (most recent call last):
  File "/usr/local/bin/pipenv", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.9/dist-packages/pipenv/vendor/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pipenv/cli/options.py", line 58, in main
    return super().main(*args, **kwargs, windows_expand_args=False)
  File "/usr/local/lib/python3.9/dist-packages/pipenv/vendor/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/pipenv/vendor/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/pipenv/vendor/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/dist-packages/pipenv/vendor/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pipenv/vendor/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pipenv/vendor/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pipenv/cli/command.py", line 233, in install
    do_install(
  File "/usr/local/lib/python3.9/dist-packages/pipenv/routines/install.py", line 170, in do_install
    do_init(
  File "/usr/local/lib/python3.9/dist-packages/pipenv/routines/install.py", line 768, in do_init
    do_lock(
  File "/usr/local/lib/python3.9/dist-packages/pipenv/routines/lock.py", line 65, in do_lock
    venv_resolve_deps(
  File "/usr/local/lib/python3.9/dist-packages/pipenv/utils/resolver.py", line 1110, in venv_resolve_deps
    c = resolve(cmd, st, project=project)
  File "/usr/local/lib/python3.9/dist-packages/pipenv/utils/resolver.py", line 985, in resolve
    raise RuntimeError("Failed to lock Pipfile.lock!")
RuntimeError: Failed to lock Pipfile.lock!
/home/USER
failed to install python modules

This installer can be resumed with:
sudo /tmp/PaperPi_rY5ya/install/install.sh
Failed during: /usr/bin/sudo /tmp/PaperPi_rY5ya/install/install.sh

**Expected behavior**
resuming installer would resume installation

**Hardware**
- Full EPD version: Waveshare Driver Hat Rev 2.3 w/ 2.13" V3
- Pi Version: Raspberry Pi 4 Model B Rev 1.5

**OS Info**
Run `cat /etc/os-release`
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

```

**`[main]` Section of Config**

**Plugin Issues**

**Additional context**
This is a relatively fresh RPI install, so I'm not sure what would be wrong. I have the Pi in a Argon40 V2 Case, which has some GPIO script components.  I had to disable a dtoverlay = GPIO23 code in /boot/config.txt for their IR receiver, but I don't think that is related. I have run the waveshare demo code via python and the screen works fine.

txoof, 2023-08-09:
@VaporwareII 
Looks like there's some dependency conflict in the Pipfile which is super odd because there aren't any pinned versions so there shouldn't be any dependency issues.

I'd like to see what's in the `/usr/local/paperpi/Pipfile` and `Pipfile.lock`

You can get them by typing:
`cat /usr/local/paperpi/Pipfile`
`cat /usr/local/paperpi/Pipfile.lock`

and then paste them into a comment.

Can you also try to run and let me know what happens?
```
$ cd /usr/local/paperpi
$ sudo rm Pipfile.lock
$ sudo pipenv install --skip-lock
```

VaporwareII, 2023-08-09:
Sure, walked away because I didn't expect you were on at the moment, lol

```
USER@Argon:~ $ cat /usr/local/paperpi/Pipfile
[[source]]
url = "https://www.piwheels.org/simple/"
verify_ssl = true
name = "piwheels"

[[source]]
url = "https://www.pypi.org/simple"
verify_ssl = true
name = "pypi"

[packages]
it8951 = {editable = true, git = "https://github.com/GregDMeyer/IT8951.git"}
epdlib = "*"
argconfigparse = "*"

[requires]
python_version = "3"

[dev-packages]
dictor = "*"
requests = "*"
pytz = "*"
feedparser = "*"
python-dateutil = "*"
currency-symbols = "*"
pycoingecko = "*"
pygal = "*"
qrcode = "*"
cairocffi = "*"
cffi ="*"
psutil = "*"
pillow = "*"
epdlib = "*"
querylms = "*"
cairosvg = "*"
ipykernel = "*"
pipreqs = "*"

```
```
USER@Argon:~ $ cat /usr/local/paperpi/Pipfile.lock
cat: /usr/local/paperpi/Pipfile.lock: No such file or directory
```

```
USER@Argon:~ $ cd /usr/local/paperpi
USER@Argon:/usr/local/paperpi $ sudo rm Pipfile.lock
rm: cannot remove 'Pipfile.lock': No such file or directory
USER@Argon:/usr/local/paperpi $ sudo pipenv install --skip-lock
The flag --skip-lock has been deprecated for removal.  Without running the lock resolver it is not possible to manage multiple package indexes.  Additionally it bypasses the build consistency guarantees provided by maintaining a lock file.
Installing dependencies from Pipfile...
An error occurred while installing epdlib! Will try again.
An error occurred while installing argconfigparse! Will try again.
Installing initially failed dependencies...
[pipenv.exceptions.InstallError]: Looking in indexes: https://www.piwheels.org/simple/, https://www.piwheels.org/simple
[pipenv.exceptions.InstallError]: Collecting epdlib (from -r /tmp/pipenv-5i6btzk4-requirements/pipenv-gwau9tww-hashed-reqs.txt (line 1))
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.6.1.3-py3-none-any.whl (43 kB)
[pipenv.exceptions.InstallError]: Collecting argconfigparse (from -r /tmp/pipenv-5i6btzk4-requirements/pipenv-gwau9tww-hashed-reqs.txt (line 2))
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/argconfigparse/ArgConfigParse-0.2.7-py3-none-any.whl (21 kB)
[pipenv.exceptions.InstallError]: INFO: pip is looking at multiple versions of epdlib to determine which version is compatible with other requirements. This could take a while.
[pipenv.exceptions.InstallError]: Collecting epdlib (from -r /tmp/pipenv-5i6btzk4-requirements/pipenv-gwau9tww-hashed-reqs.txt (line 1))
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.7-py3-none-any.whl (40 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.5-py3-none-any.whl (39 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.3-py3-none-any.whl (39 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.2-py3-none-any.whl (41 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.1-py3-none-any.whl (38 kB)
[... 558 lines cut ...]

[requires]
python_version = "3"

[dev-packages]
dictor = "*"
requests = "*"
pytz = "*"
feedparser = "*"
python-dateutil = "*"
currency-symbols = "*"
pycoingecko = "*"
pygal = "*"
qrcode = "*"
cairocffi = "*"
cffi ="*"
psutil = "*"
pillow = "*"
epdlib = "*"
querylms = "*"
cairosvg = "*"
ipykernel = "*"
pipreqs = "*"

```

harperreed, 2023-08-09:
FWIW i am having the exact same issue. 

```shell
harper@raspberrypi:/usr/local/paperpi $ sudo pipenv install --skip-lock
The flag --skip-lock has been deprecated for removal.  Without running the lock resolver it is not possible to manage multiple package indexes.  Additionally it bypasses the build consistency guarantees provided by maintaining a lock file.
Installing dependencies from Pipfile...
An error occurred while installing epdlib! Will try again.
An error occurred while installing argconfigparse! Will try again.
Installing initially failed dependencies...
[pipenv.exceptions.InstallError]: Looking in indexes: https://www.piwheels.org/simple/, https://www.piwheels.org/simple
[pipenv.exceptions.InstallError]: Collecting epdlib (from -r /tmp/pipenv-815hne9c-requirements/pipenv-uvv0lg9p-hashed-reqs.txt (line 1))
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.6.1.3-py3-none-any.whl (43 kB)
[pipenv.exceptions.InstallError]: Collecting argconfigparse (from -r /tmp/pipenv-815hne9c-requirements/pipenv-uvv0lg9p-hashed-reqs.txt (line 2))
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/argconfigparse/ArgConfigParse-0.2.7-py3-none-any.whl (21 kB)
[pipenv.exceptions.InstallError]: INFO: pip is looking at multiple versions of epdlib to determine which version is compatible with other requirements. This could take a while.
[pipenv.exceptions.InstallError]: Collecting epdlib (from -r /tmp/pipenv-815hne9c-requirements/pipenv-uvv0lg9p-hashed-reqs.txt (line 1))
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.7-py3-none-any.whl (40 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.5-py3-none-any.whl (39 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.3-py3-none-any.whl (39 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.2-py3-none-any.whl (41 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3.1-py3-none-any.whl (38 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.3-py3-none-any.whl (38 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.4.2-py3-none-any.whl (41 kB)
[pipenv.exceptions.InstallError]: INFO: pip is still looking at multiple versions of epdlib to determine which version is compatible with other requirements. This could take a while.
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.2.1-py3-none-any.whl (37 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.1.5-py3-none-any.whl (118 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.1.4-py3-none-any.whl (117 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.1.2-py3-none-any.whl (37 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.0.1.6-py3-none-any.whl (36 kB)
[pipenv.exceptions.InstallError]: INFO: This is taking longer than usual. You might need to provide the dependency resolver with stricter constraints to reduce runtime. See https://pip.pypa.io/warnings/backtracking for guidance. If you want to abort this run, press Ctrl + C.
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.0.1.5-py3-none-any.whl (23 kB)
[pipenv.exceptions.InstallError]:   Using cached https://www.piwheels.org/simple/epdlib/epdlib-0.0.1.2-py3-none-any.whl (35 kB)
[pipenv.exceptions.InstallError]:
[pipenv.exceptions.InstallError]: The conflict is caused by:
[pipenv.exceptions.InstallError]:     epdlib 0.6.1.3 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.4.3.7 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.4.3.5 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.4.3.3 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.4.3.2 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.4.3.1 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.4.3 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.4.2 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.2.1 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.1.5 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.1.4 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.1.2 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.0.1.6 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.0.1.5 depends on Pillow
[pipenv.exceptions.InstallError]:     epdlib 0.0.1.2 depends on Pillow
[pipenv.exceptions.InstallError]:
[pipenv.exceptions.InstallError]: To fix this you could try to:
[pipenv.exceptions.InstallError]: 1. loosen the range of package versions you've specified
[pipenv.exceptions.InstallError]: 2. remove package versions to allow pip attempt to solve the dependency conflict
[pipenv.exceptions.InstallError]: ERROR: Cannot install -r /tmp/pipenv-815hne9c-requirements/pipenv-uvv0lg9p-hashed-reqs.txt (line 1) because these package versions have conflicting dependencies.
[... cut ...]
Successfully built RPi.GPIO spidev
Installing collected packages: spidev, RPi.GPIO, Pillow, epdlib
Successfully installed Pillow-10.0.0 RPi.GPIO-0.7.1 epdlib-0.6.1.3 spidev-3.6
```

and then inspect it via `pipenv graph`:

```shell
harper@raspberrypi:/usr/local/paperpi $ pipenv graph
epdlib==0.6.1.3
├── Pillow [required: Any, installed: 10.0.0]
├── RPi.GPIO [required: Any, installed: 0.7.1]
└── spidev [required: Any, installed: 3.6]
IT8951==0.1.1
```

I wasn't able to get much further. Will update as i keep hacking

harperreed, 2023-08-09:
I was able to install the dependencies manually using `pipenv run pip <dependency>` and paperpi.py now executes.

VaporwareII, 2023-08-10:
Not sure if what I did fixed it, because it still won't lock normally and I have never used pip before.  the command sudo pipenv install fails with the normal dependency error.  The remote install script also still fails.

I played with it a bit today and got the --skip-lock install to finish. Does that mean I can run it?  I don't know what the next step is, because I do not know how to run the program.

```

USER@Argon:/usr/local/paperpi $ sudo pipenv install --skip-lock
Installing dependencies from Pipfile...
To activate this project's virtualenv, run pipenv shell.
Alternatively, run a command inside the virtualenv with pipenv run.

```
The things I tried are what harperreed presented above, where you 
```
sudo pipenv run pip install epdlib
sudo pipenv run pip install argconfigparse
sudo pipenv run pip install IT8951

```
and run graph after

```
USER@Argon:/usr/local/paperpi $ pipenv graph
ArgConfigParse==0.2.7
epdlib==0.6.1.3
├── Pillow [required: Any, installed: 10.0.0]
├── RPi.GPIO [required: Any, installed: 0.7.1]
└── spidev [required: Any, installed: 3.6]
IT8951==0.1.1
```

I found the pypi notes section https://pypi.org/project/epdlib/ about the waveshare-epd library and saw it didn't install to /src.  /usr/local/paperpi/venv/src only had the it8951 folder.  I used the command in the pypi notes to install it.  Now the dependency and directory structure looks like the below.

```
USER@Argon:/usr/local/paperpi $ pipenv graph
ArgConfigParse==0.2.7
epdlib==0.6.1.3
├── Pillow [required: Any, installed: 10.0.0]
├── RPi.GPIO [required: Any, installed: 0.7.1]
└── spidev [required: Any, installed: 3.6]
IT8951==0.1.1
waveshare-epd==0.0.0
├── Pillow [required: Any, installed: 10.0.0]
├── RPi.GPIO [required: Any, installed: 0.7.1]
└── spidev [required: Any, installed: 3.6]
```
```
USER@Argon:/usr/local/paperpi $ tree -d
.
├── config
│   ├── logging.cfg
│   ├── paperpi_base.ini
│   └── paperpi.ini
├── debian_packages-paperpi.txt
├── fonts
├── library
│   ├── CacheFiles.py
│   ├── get_help.py
│   ├── __init__.py
│   ├── InterruptHandler.py
│   ├── Plugin.py
│   ├── PluginTools.py
│   └── run_module.py
├── my_constants.py
├── paperpi.py
├── Pipfile
├── plugins
├── .venv
│   ├── bin
│   │   ├── activate
│   │   ├── activate.csh
│   │   ├── activate.fish
│   │   ├── activate.nu
│   │   ├── activate.ps1
│   │   ├── activate_this.py
│   │   ├── pip
│   │   ├── pip3
│   │   ├── pip-3.9
│   │   ├── pip3.9
│   │   ├── python -> /usr/bin/python3.9
│   │   ├── python3 -> python
│   │   ├── python3.9 -> python
│   │   ├── wheel
│   │   ├── wheel3
│   │   ├── wheel-3.9
│   │   └── wheel3.9
│   ├── .gitignore
│   ├── lib
│   │   └── python3.9
│   │       └── site-packages
│   │           ├── ArgConfigParse
│   │           ├── ArgConfigParse-0.2.7.dist-info
│   │           ├── _distutils_hack
│   │           ├── epdlib
│   │           ├── epdlib-0.6.1.3.dist-info
│   │           ├── PIL
│   │           ├── Pillow-10.0.0.dist-info
│   │           ├── Pillow.libs
│   │           ├── pip
│   │           ├── pip-23.2.1.dist-info
│   │           ├── pkg_resources
│   │           ├── RPi
│   │           ├── RPi.GPIO-0.7.1.dist-info
│   │           ├── setuptools
│   │           ├── setuptools-68.0.0.dist-info
│   │           ├── spidev-3.6.dist-info
│   │           ├── wheel
│   │           └── wheel-0.41.0.dist-info
│   ├── .project
│   ├── pyvenv.cfg
│   └── src
│       ├── it8951
│       │   ├── build
│       │   ├── .git
│       │   ├── .gitignore
│       │   ├── IT8951
│       │   ├── IT8951.egg-info
│       │   ├── setup.py
│       │   └── test
│       └── waveshare-epd
│           ├── Arduino
│           ├── .git
│           ├── RaspberryPi_JetsonNano
│           ├── README.md
│           ├── STM32
└── waveshare_epd
```

txoof, 2023-08-10:
Wow! Good work fiddling! I'm not sure what caused the avalanche of piwheels errors. It's good info though. I'll have to investigate that further.

Thanks for all the good info. I wish all bug reports were this thorough!

Running it is a little tricky because the config file, entry script and unit files were not installed. You can get around that though.

You can try the following:

From the paperpi directory run `pipenv run paperpi.py`

That should shove a config file in `~/.config/txoof.paperpi.ini`

Edit that to configure your epd and enable a few plugins. Then run `pipenv run paperpi.py` again and it **should** start up.

Tomorrow I'll try to diagnose the problem and see if I can recreate it.

One more useful bit of info, can you run `which python3` and `python3 -V` ?

I'm wondering if you have a weird python environment that might be causing strange behavior.

VaporwareII, 2023-08-10:
I'm sick with covid so I don't have much else to do and troubleshooting is about all I can focus on.

```
USER@Argon:/usr/local/paperpi $ which python3
/usr/bin/python3
USER@Argon:/usr/local/paperpi $ python3 -V
Python 3.9.2
USER@Argon:/usr/local/paperpi $ pipenv run paperpi.py
Error: the command paperpi.py could not be found within PATH or Pipfile's [scripts].

```

When you say run that command in the paperpi folder do you mean /local/usr/bin/paperpi? I assume not the ~/USER/paperpi one because there's just an install script there.  Maybe I accidentally put that one there when I was trying to figure out how to download from github.

VaporwareII, 2023-08-10:
Hmm, it seems to work with an explicit path to the .py

```
USER@Argon:/usr/local/paperpi $ pipenv run /usr/local/paperpi/paperpi.py
Traceback (most recent call last):
  File "/usr/local/paperpi/paperpi.py", line 37, in <module>
    from library.Plugin import Plugin
  File "/usr/local/paperpi/library/__init__.py", line 1, in <module>
    from .CacheFiles import CacheFiles
  File "/usr/local/paperpi/library/CacheFiles.py", line 13, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
USER@Argon:/usr/local/paperpi $ pipenv shell
Launching subshell in virtual environment...
USER@Argon:/usr/local/paperpi $  . /usr/local/paperpi/.venv/bin/activate
(paperpi) USER@Argon:/usr/local/paperpi $ ls
config  debian_packages-paperpi.txt  fonts  library  my_constants.py  paperpi.py  Pipfile  Pipfile.lock  plugins  tree.txrt  tree.txt  waveshare_epd
(paperpi) USER@Argon:/usr/local/paperpi $ python3 paperpi.py
Traceback (most recent call last):
  File "/usr/local/paperpi/paperpi.py", line 37, in <module>
    from library.Plugin import Plugin
  File "/usr/local/paperpi/library/__init__.py", line 1, in <module>
    from .CacheFiles import CacheFiles
  File "/usr/local/paperpi/library/CacheFiles.py", line 13, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'

txoof, 2023-08-11:
@VaporwareII 

After all that, does it run?

I'll need to review this on a bigger screen when I have access to my development environment.

I really appreciate all the debugging and hacking you all did. 

I can't see anything weird in your environment, so I'm really curious to get to the bottom of this. 

@harperreed your comments are super helpful too. I think the root cause is that pillow is failing to install from  Piwheels and that's cascading down into epdlib failing. It looks like when you manually installed epdlib, it was pulled from pypi instead.

VaporwareII, 2023-08-11:
Yeah I took piwheels out of the Pipfile entirely because I saw a comment in the pipenv github that it gets confused between two sources sometimes.  but I wasn't properly debugging by trying ust one change at a time, I did that and reformat the pipfile too.  There was another comment that it checks top to bottom so I tried moving the packages around.

It runs.  I chose not to use daemon mode so I only see a few lines, and I chose the weather plugin so the data doesn't really change between when I glance at it.

```
USER@Argon:/usr/local/paperpi $ paperpi

######################################################################
This appears to be the first time PaperPi has been run.
    A user configuration file created: /home/USER/.config/com.txoof.paperpi/paperpi.ini
    At minimum you edit this file and add a display_type and enable one plugin.

    Edit the configuration file with:
       $ nano /home/USER/.config/com.txoof.paperpi/paperpi.ini
######################################################################

***Exiting***
USER@Argon:/usr/local/paperpi $ nano /home/USER/.config/com.txoof.paperpi/paperpi.ini
USER@Argon:/usr/local/paperpi $ sudo nano /etc/default/paperpi.ini
USER@Argon:/usr/local/paperpi $ paperpi
21:06:14 paperpi :get_config_files:256 :INFO       - using configuration files to configure PaperPi: [PosixPath('/usr/local/paperpi/config/paperpi.ini'), PosixPath('/home/USER/.config/com.txoof.paperpi/paperpi.ini')]
21:06:14 paperpi :main      :658 :INFO       - ********** PaperPi 0.5.4.0 RGB Starting **********
21:06:14 paperpi :main      :662 :INFO       - PaperPi is running in on-demand mode
21:06:15 Block   :padding   :346 :WARNING    - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
21:06:15 Block   :padding   :353 :WARNING    - the padded area available may be too small to display any content: Area: (250, 12), Padded Area: [230, -8]
21:06:15 Block   :font_size :824 :WARNING    - no font size set, using {font_size}
21:06:22 Block   :padding   :346 :WARNING    - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
21:06:22 Block   :padding   :353 :WARNING    - the padded area available may be too small to display any content: Area: (150, 12), Padded Area: [130, -8]
21:06:22 Block   :font_size :824 :WARNING    - no font size set, using {font_size}
21:06:22 Block   :padding   :346 :WARNING    - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
21:06:22 Block   :padding   :353 :WARNING    - the padded area available may be too small to display any content: Area: (100, 12), Padded Area: [80, -8]
21:06:22 Block   :font_size :824 :WARNING    - no font size set, using {font_size}
^CUSER@Argon:/usr/local/paperpi $ paperpi
07:28:55 paperpi :get_config_files:256 :INFO       - using configuration files to configure PaperPi: [PosixPath('/usr/local/paperpi/config/paperpi.ini'), PosixPath('/home/USER/.config/com.txoof.paperpi/paperpi.ini')]
07:28:55 paperpi :main      :658 :INFO       - ********** PaperPi 0.5.4.0 RGB Starting **********
07:28:55 paperpi :main      :662 :INFO       - PaperPi is running in on-demand mode
07:28:55 Block   :padding   :346 :WARNING    - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
07:28:55 Block   :padding   :353 :WARNING    - the padded area available may be too small to display any content: Area: (250, 12), Padded Area: [230, -8]
07:28:55 Block   :font_size :824 :WARNING    - no font size set, using {font_size}
07:29:03 Block   :padding   :346 :WARNING    - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
07:29:03 Block   :padding   :353 :WARNING    - the padded area available may be too small to display any content: Area: (150, 12), Padded Area: [130, -8]
07:29:03 Block   :font_size :824 :WARNING    - no font size set, using {font_size}
07:29:03 Block   :padding   :346 :WARNING    - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
07:29:03 Block   :padding   :353 :WARNING    - the padded area available may be too small to display any content: Area: (100, 12), Padded Area: [80, -8]
07:29:03 Block   :font_size :824 :WARNING    - no font size set, using {font_size}
```

txoof, 2023-08-11:
@VaporwareII 
> Hmm, looks like you don't need the waveshare-epd line in the Pipfile since it's not a python package. I took it out and it runs all the way through the installer. edit: also could have been a typo. it should have been waveshare_edp as the package name.

How on earth did that waveshare line get in there? I can't figure out how that got into the dependencies at all. The waveshare libraries should be distributed as part of the source.

txoof, 2023-08-11:
@VaporwareII and @harperreed
I'm trying to reproduce the errors you all are getting and I'm just not getting anywhere with it. I just created a fresh image from Bullseye32 lite and tried to break the installer, but can't reproduce the errors you're seeing. 

Would either of you all be willing to take an image of your SD card and send it to me? I'd love to poke at your setups and see if I can figure out what's going on.

**BE AWARE:**
I'll be able to see:
- your password hash
- anything in your /home folders
- any config files in /etc/

I promise not to do anything awful with it, but I just want you to be aware of what you're sharing before you do it...

VaporwareII, 2023-08-11:
> @VaporwareII
>
>
> How on earth did that waveshare line get in there? I can't figure out how that got into the dependencies at all. The waveshare libraries should be distributed as part of the source.

Oh I put it in there as part of my debugging.  throwing things at the wall.  I mentioned it in comment #6

my uname --all 
`Linux Argon 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

I'm on 64bit, maybe that's related?  I'll think about the image, I don't think I have spent much time configuring this box, but it's on a SSD instead of SD card.

also I noticed after the script finished the first part where it fails, the Pipfile gets a lot longer after it merges requirements.txt

```
[[source]]
url = "https://www.pypi.org/simple"
verify_ssl = true
name = "pypi"

[requires]
python_version = '3.9'

[packages]
IT8951 = {git = "https://github.com/GregDMeyer/IT8951.git", editable = true}
epdlib = "*"
ArgConfigParse = '>=0.2.7'
dictor = "*"
requests = "*"
querylms = "*"
cairocffi = "*"
cffi = "*"
cairosvg = "*"
currency-symbols = "*"
feedparser = "*"
pycoingecko = "*"
pygal = "*"
qrcode = "*"
pillow = "*"
python-dateutil = "*"
psutil = "*"
pytz = "*"

[dev-packages]
dictor = "*"
requests = "*"
pytz = "*"
feedparser = "*"
python-dateutil = "*"
currency-symbols = "*"
pycoingecko = "*"
pygal = "*"
qrcode = "*"
cairocffi = "*"
cffi ="*"
psutil = "*"
pillow = "*"
epdlib = "*"
querylms = "*"
cairosvg = "*"
ipykernel = "*"
pipreqs = "*"
```

txoof, 2023-08-12:
> I'm on 64bit, maybe that's related?

Yes! That's definitely related! The piwheels packages are built against a 32 bit arch! That's why epdlib fails miserably, it's trying to install a 32bit version of Pillow and that's not available.

I'll see if I can reproduce it with a 64bit version of the OS.

At minimum, I can add that requirement to the docs and probably check for it in the installer.

Good thinking!

txoof, 2023-08-13:
@VaporwareII and @harperreed,

I've updated the install scripts to check for 64 bit systems; the remote installer/installer will both abort if they detect a 64 bit system. You can test out the new installer here and see if it fails on your systems:

`curl -fsSL https://raw.githubusercontent.com/txoof/PaperPi/main/install/remote_install.sh | bash -s -- -b I120_installer_fails`

Would you mind checking the readme testing the [manual install instructions](https://github.com/txoof/PaperPi/tree/I120_installer_fails#manual-install-for-rpios-64-bit) I've worked up for 64 bit systems?

Thanks again for your troubleshooting and help figuring out what was going on.

txoof, 2023-08-18:
@VaporwareII and @harperreed 

I'm going to close this ticket. If you have more to add, we can reopen it.

closed 2023-08-18 by txoof