txoof/PaperPi #123: Create a web interface for configuration & other features

txoof, 2023-08-20:
**Is your feature request related to a problem? Please describe.**

Editing config files isn't super friendly. Previewing plugins and layouts is not possible.

**Describe the solution you'd like**

Create a web interface like this one: https://github.com/tlstommy/PiInk/tree/main

9and3r, 2023-09-03:
- Decide Icon for the web (Favicon at least) and main colors (used on buttons)
- Decide if web source should be on main repo
- Look at the timeout for plugins (It breaks the preview on web. Thread related)
- Extract build_plugins_list function to use it only on one of the plugins
- @9and3r (if a plugin has configurable: false should not appear on the config
- Document allowed types in plugin config: "string", "bool", "float", "int" ("choose" property)

9and3r, 2023-09-10:
> * [ ]  Decide Icon for the web (Favicon at least) and main colors (used on buttons)
> * [ ]  Decide if web source should be on main repo
> * [ ]  Look at the timeout for plugins (It breaks the preview on web. Thread related) @txoof
> * [ ]  Extract build_plugins_list function to use it only on one of the plugins @ txoof
> * [x]  @9and3r (if a plugin has configurable: false should not appear on the config
>   Document allowed types in plugin config: "string", "bool", "float", "int" ("choose" property)

Branch: https://github.com/9and3r/PaperPi/tree/web_configuration

txoof, 2023-09-13:
@9and3r Can you give me some more examples of how the `plugin_timeout` breaks when you're configuring a plugin through the web interface?

I can't think of a reason it would die due to the timeout.

9and3r, 2023-09-13:
It is not getting to the timeout. The problem is the signal library being called from another thread.

The server is running on another thread, so it calls the update function from outside the main thread.
The signal cannot be set outside of main thread according to the doc.

[https://docs.python.org/3/library/signal.html](https://docs.python.org/3/library/signal.html)
`Besides, only the main thread of the main interpreter is allowed to set a new signal handler.`

9and3r, 2023-09-13:
Example python file to test this. The first runSignal will work. The second part will fail because it runs in a new thread
```
import signal
import time
from threading import Thread

def _alarm_handler(signum, frame):
    print("Alarm called")

def runSignal():
    signal.signal(signal.SIGALRM, _alarm_handler)
    signal.alarm(1)
    time.sleep(3)

# This will work
runSignal()

# This will fail
t = Thread(target=runSignal)
t.start()

time.sleep(3)
```

txoof, 2023-09-17:
- [ ] add option disable signal to Plugin() class
- [ ] refactor the configuration functions into new library

txoof, 2023-09-17:
- [ ] update installer to install web_static
- [ ] script to build web_static from source

txoof, 2023-09-17:
PR #136 resolves `add option disable signal`

txoof, 2023-09-17:
EpdLib still needs to be forced to recalculate all the layout dimensions. There's still an underlying bug.

txoof, 2023-09-17:
refactor config functions into new module in PR #138

txoof, 2023-09-17:
Recipe for testing out the new configuration bits
```
# get the command line arguments 
cla = get_cmd_line_args()

# get the configuration files - this includes a list of the config files that were ingested
cfg = get_config_files(cla)

# parse the config files 
pcfg = parse_config(cfg)

# create a plugin using the main config and a single plugin from the list of configured plugins
mp = configure_plugin(main_cfg=pcfg['main'], config=pcfg['plugins'][0], resolution=(800,600), cache=CacheFiles())
```

txoof, 2023-09-23:
- [ ] Add functions to display user-facing functions for a plugin
- [ ] Add function for calling user-facing functions and returning a string
@9and3r -- anything else?

closed 2026-11-09 by txoof-bot