txoof/PaperPi #124: Problem installing development enviroment

9and3r, 2023-08-20:
**Describe the bug**
Could not set development enviroment

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Hardware**
- Full EPD version: 
- Raspberry Pi Zero W Rev 1.1

**OS Info**
Run `cat /etc/os-release`
```
Raspbian GNU/Linux 11 (bullseye)
```

**`[main]` Section of Config**
Copy and paste the `[main]` section of your config file
- Daemon mode (start at boot) use: `/etc/default/paperpi.ini`
- Single user mode (start manually) use: `~/.config/com.txoof.paperpi/paperpi.ini`

```
paste here
```
[PaperPi log.txt](https://github.com/txoof/PaperPi/files/12388837/PaperPi.log.txt)

**Plugin Issues**
Copy and paste the `[Plugin: Your Plugin]` section of your config file
- Daemon mode (start at boot) use: `/etc/default/paperpi.ini`
- Single user mode (start manually) use: `~/.config/com.txoof.paperpi/paperpi.ini`

```
paste here
```

**Additional context**
Add any other context about the problem here.

txoof, 2023-08-20:
Cleaning development modules fails with the errors below
```
pi@gewoon:~/src/PaperPi/utilities $ ./create_devel_environment.sh  -j
checking if SPI is enabled

SPI OK
checking for required debian packages
required packages installed
removing all previous development modules
Creating a virtualenv for this project...
Pipfile: /home/pi/src/Pipfile
Using default python from /usr/bin/python3 (3.9.2) to create virtualenv...
⠧ Creating virtual environment...created virtual environment CPython3.9.2.final.0-32 in 1657ms
  creator CPython3Posix(dest=/home/pi/.local/share/virtualenvs/src-Xw-q2ltK, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/pi/.local/share/virtualenv)
    added seed packages: pip==23.2.1, setuptools==68.0.0, wheel==0.41.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator

✔ Successfully created virtual environment!
Virtualenv location: /home/pi/.local/share/virtualenvs/src-Xw-q2ltK
Creating a Pipfile for this project...
Un-installing [dev-packages]...
Traceback (most recent call last):
  File "/home/pi/.local/bin/pipenv", line 8, in <module>
    sys.exit(cli())
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/vendor/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/cli/options.py", line 58, in main
    return super().main(*args, **kwargs, windows_expand_args=False)
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/vendor/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/vendor/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/vendor/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/vendor/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/vendor/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/vendor/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/vendor/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/cli/command.py", line 286, in uninstall
    retcode = do_uninstall(
  File "/home/pi/.local/lib/python3.9/site-packages/pipenv/routines/uninstall.py", line 83, in do_uninstall
    preserve_packages |= set(project_pkg_names[category])
KeyError: 'default'
installing all plugin development requirements
Installing development dependencies for all plugins:
dictor
pytz
requests

feedparser
requests
currency_symbols
dictor
feedparser
pycoingecko
pygal
qrcode
requests
cairocffi
cffi
CairoSVG
Pillow

Pillow
dictor
requests
dictor
epdlib
requests
psutil
dictor
Pillow
python_dateutil
requests
epdlib
QueryLMS
requests

Pillow
requests

tempfile: /tmp/tmp.13fPQnIXwH
Creating a virtualenv for this project...
Pipfile: /home/pi/src/PaperPi/Pipfile
Using /usr/bin/python3.9 (3.9.2) to create virtualenv...
⠦ Creating virtual environment...created virtual environment CPython3.9.2.final.0-32 in 1525ms
  creator CPython3Posix(dest=/home/pi/.local/share/virtualenvs/PaperPi-VBShxqF-, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/pi/.local/share/virtualenv)
    added seed packages: pip==23.2.1, setuptools==68.0.0, wheel==0.41.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator

✔ Successfully created virtual environment!
Virtualenv location: /home/pi/.local/share/virtualenvs/PaperPi-VBShxqF-
Requirements file provided! Importing into Pipfile...
Installing dependencies from Pipfile.lock (5d562c)...
Installing dependencies from Pipfile.lock (5d562c)...
To activate this project's virtualenv, run pipenv shell.
Alternatively, run a command inside the virtualenv with pipenv run.
adding kernel spec: PaperPi-VBShxqF-
Installed kernelspec PaperPi-VBShxqF- in /home/pi/.local/share/jupyter/kernels/paperpi-vbshxqf-
```

txoof, 2023-08-20:
development script should check for/install jupyter.

txoof, 2023-08-20:
cleaning development modules fails

txoof, 2023-08-21:
make sure jupyter is installed

closed 2023-08-31 by txoof