txoof/PaperPi #141: Not working with 7.8inch e-Paper HAT (IT8951)

Hoerli1337, 2023-10-10:
Hi!
I get the following error message when I try to start PaperPi:
```
# paperpi -d
21:55:58 paperpi :get_config_files:256 :INFO       - using configuration files to configure PaperPi: [PosixPath('/usr/local/paperpi/config/paperpi.ini'), PosixPath('/etc/default/paperpi.ini')]
21:55:58 paperpi :main      :658 :INFO       - ********** PaperPi 0.5.4.1 RGB Starting **********
21:55:58 paperpi :main      :660 :INFO       - PaperPi is running in daemon mode
Traceback (most recent call last):
  File "/usr/local/paperpi/paperpi.py", line 738, in <module>
    exit_code = main()
  File "/usr/local/paperpi/paperpi.py", line 673, in main
    screen_return = setup_display(config)
  File "/usr/local/paperpi/paperpi.py", line 435, in setup_display
    screen = Screen(epd=epd, vcom=vcom)
  File "/usr/local/paperpi/.venv/lib/python3.9/site-packages/epdlib/Screen.py", line 234, in __init__
    self.epd = epd
  File "/usr/local/paperpi/.venv/lib/python3.9/site-packages/epdlib/Screen.py", line 45, in new_f
    return f(self, *args, **kwds)
  File "/usr/local/paperpi/.venv/lib/python3.9/site-packages/epdlib/Screen.py", line 324, in epd
    myepd = self._load_hd(epd)
  File "/usr/local/paperpi/.venv/lib/python3.9/site-packages/epdlib/Screen.py", line 430, in _load_hd
    myepd = AutoEPDDisplay(vcom=self.vcom)
  File "/usr/local/paperpi/.venv/src/it8951/IT8951/display.py", line 224, in __init__
    epd = EPD(vcom=vcom, bus=bus, device=device, data_hz=spi_hz)
  File "/usr/local/paperpi/.venv/src/it8951/IT8951/interface.py", line 33, in __init__
    self.update_system_info()
  File "/usr/local/paperpi/.venv/src/it8951/IT8951/interface.py", line 103, in update_system_info
    raise RuntimeError("communication with device failed")
RuntimeError: communication with device failed
```

- Raspberry Pi 3B+
- Raspbian OS
- Freshly installed OS
- No plugins enabled, only display type and vcom specified
- SPI is enabled

txoof, 2023-10-12:
@Hoerli1337 

What screen are you using? I assume it's an HD screen, but I just want to be certain.

Does the screen work with any of the demos from WaveShare?

Can you also share the ini file you're using? The `[main]' section should be enough.

Hoerli1337, 2023-10-12:
Okay that was weird.
It runs now.
But a reboot was not the solution, I had done several even.
I had to disconnect the Pi completely from the power for a short time, then everything started cleanly automatically.
[image]

I have been trying to get Paperpi to run, as unfortunately I have had no success with your driver either.
My goal is to display some data from my openHAB instance.
For this I wanted to create a small Python script. But I am still completely at the beginning.

txoof, 2023-10-17:
did you eventually get PaperPi to run?
It looks like you are running on Bookworm32 with Python 11?

Hoerli1337, 2023-11-05:
Sorry for the late response, unfortunately I had other things to do.
Yes PaperPi runs with Python 11 without any problems after I solved the problem with the driver.

txoof, 2023-11-06:
Great news! Thanks for the update.

MehdyShinwari, 2024-03-19:
Hi everyone!
I have a similar issue. Rebooting or unplugging the pi from power did not solve the issue. My setup:

- RPi 4B 4GB
- Raspbian OS (bullseye)
- SPI is enabled
- Clock plugin enabled, display type (HD) and vcom specified (-1.97)
- 7.8inch e-Paper HAT (IT8951(B))
- Waveshare Demo works 

Error Log:
pi@raspberrypi:~ $ sudo /usr/local/bin/paperpi -d 
14:11:37 paperpi :get_config_files:235 :INFO       - using configuration files to configure PaperPi: [PosixPath('/usr/local/paperpi/config/paperpi.ini'), PosixPath('/etc/default/paperpi.ini')]
14:11:37 paperpi :main      :655 :INFO       - ********** PaperPi 0.5.6.0 RGB Starting **********
14:11:37 paperpi :main      :657 :INFO       - PaperPi is running in daemon mode
14:11:37 paperpi :main      :663 :DEBUG      - configuration:
{'main': {'display_type': 'HD', 'vcom': -1.97, 'max_refresh': 4, 'log_level': 'DEBUG', 'splash': True, 'rotation': 0, 'mirror': False, 'color': False, 'no_wipe': True, 'plugin_timeout': 30, 'daemon': True}, 'Plugin: default fallback plugin': {'layout': 'layout', 'plugin': 'default', 'refresh_rate': 30, 'min_display_time': 60, 'max_priority': 32768}, 'xPlugin: Basic Clock': {'layout': 'layout', 'plugin': 'basic_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'xPlugin: decimal binary clock': {'layout': 'layout', 'plugin': 'dec_bin_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'xPlugin: default fallback plugin': {'layout': 'layout', 'plugin': 'default', 'refresh_rate': 30, 'min_display_time': 60, 'max_priority': 32768}, 'xPlugin: A Demo Plugin': {'layout': 'layout', 'plugin': 'demo_plugin', 'min_display_time': 30, 'max_priority': 1, 'your_name': 'Slartybartfast', 'your_color': 'chartreuse'}, 'xPlugin: Librespot': {'layout': 'layout', 'plugin': 'librespot_client', 'refresh_rate': 10, 'max_priority': 0, 'min_display_time': 15, 'player_name': 'LibreSpot-Spotify', 'idle_timeout': 10, 'port': 24879}, 'xPlugin: LMS - Your Player Name': {'layout': 'layout', 'plugin': 'lms_client', 'player_name': 'Your Player Name', 'refresh_rate': 5, 'min_display_time': 30, 'max_priority': 0, 'idle_timeout': 15}, 'xPlugin: Weather Adis Ababa': {'layout': 'layout', 'plugin': 'met_no', 'refresh_rate': 300, 'min_display_time': 50, 'max_priority': 2, 'location_name': 'Adis Ababa', 'lat': 9.0, 'lon': 38.75, 'email': 'you@host.diamond'}, 'xPlugin: Moon Phase': {'layout': 'layout', 'plugin': 'moon_phase', 'min_display_time': 30, 'max_priority': 2, 'refresh_rate': 1200, 'email': 'you@host.diamond', 'location_name': 'Europe/Amsterdam', 'lat': 52.3, 'lon': 4.9}, 'xPlugin: New Yorker Comic': {'layout': 'layout', 'plugin': 'newyorker', 'day_range': 5, 'refresh_rate': 120, 'min_display_time': 60, 'max_priority': 2}, 'xPlugin: Reddit Quotes': {'layout': 'layout', 'plugin': 'reddit_quote', 'refresh_rate': 100, 'min_display_time': 50, 'max_priority': 2, 'max_length': 144}, 'xPlugin: Slideshow': {'layout': 'layout', 'plugin': 'slideshow', 'refresh_rate': 90, 'min_display_time': 50, 'max_priority': 2, 'image_path': '/pi/documents/images', 'order': 'random', 'frame': 'black & silver: matted'}, 'xPlugin: System Info': {'layout': 'layout', 'plugin': 'system_info', 'min_display_time': 45, 'max_priority': 2, 'storage_unit': 'GB', 'text_color': 'BLUE', 'bkground_color': 'WHITE'}, 'xPlugin: Word Clock': {'layout': 'layout', 'plugin': 'word_clock', 'refresh_rate': 125, 'min_display_time': 255, 'max_priority': 2}, 'xPlugin: XKCD Comic Plugin': {'layout': 'layout', 'plugin': 'xkcd_comic', 'refresh_rate': 1200, 'min_display_time': 120, 'max_x': 800, 'max_y': 600, 'resize': 0, 'max_retries': 10, 'max_priority': 2}, 'Plugin: Basic Clock': {'layout': 'layout', 'plugin': 'basic_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, '__cmd_line': {'compatible': False, 'list_plugins': False}}

14:11:37 Screen  :_load_hd  :429 :DEBUG      - configuring IT8951 epd
Traceback (most recent call last):
  File "/usr/local/paperpi/paperpi.py", line 746, in <module>
    exit_code = main()
  File "/usr/local/paperpi/paperpi.py", line 668, in main
    screen_return = setup_display(config)
  File "/usr/local/paperpi/paperpi.py", line 407, in setup_display
    screen = Screen(epd=epd, vcom=vcom)
  File "/usr/local/paperpi/venv_paperpi/lib/python3.9/site-packages/epdlib/Screen.py", line 240, in __init__
    self.epd = epd
  File "/usr/local/paperpi/venv_paperpi/lib/python3.9/site-packages/epdlib/Screen.py", line 63, in new_f
    return f(self, *args, **kwds)
  File "/usr/local/paperpi/venv_paperpi/lib/python3.9/site-packages/epdlib/Screen.py", line 330, in epd
    myepd = self._load_hd(epd)
  File "/usr/local/paperpi/venv_paperpi/lib/python3.9/site-packages/epdlib/Screen.py", line 436, in _load_hd
    myepd = AutoEPDDisplay(vcom=self.vcom)
  File "/usr/local/paperpi/venv_paperpi/lib/python3.9/site-packages/IT8951/display.py", line 215, in __init__
    epd = EPD(vcom=vcom, bus=bus, device=device, data_hz=spi_hz)
  File "/usr/local/paperpi/venv_paperpi/lib/python3.9/site-packages/IT8951/interface.py", line 35, in __init__
    self.update_system_info()
  File "/usr/local/paperpi/venv_paperpi/lib/python3.9/site-packages/IT8951/interface.py", line 105, in update_system_info
    raise RuntimeError("communication with device failed")
RuntimeError: communication with device failed

Any help is appreciated, thanks!

txoof, 2024-03-19:
@MehdyShinwari 

Sorry it's not working for you?

Then can you try running the following commands and providing the output from each:
* `cat /etc/os-release` ?
* `/usr/local/bin/paperpi -c /etc/default/init.d` -- provide the full OS info
* `ls -l /dev/spidev*` -- check SPI status
* `groups paperpi`
* `groups YOUR_USER_NAME`

I have a couple ideas of the problem, but need more information to move forward.

**hint:** You can surround your output with three back tick characters "```" before and after the output. The back tick characters turns word wrapped text into a code block

MehdyShinwari, 2024-03-19:
So I installed PaperPi on a fresh install of bookworm... and it worked without any issues! Seems that the problem was really bullseye.
Thank you very much!
Have a nice day!

txoof, 2024-03-20:
@MehdyShinwari I'll call that out in the README explicitly. The README still claims it works with Buster which is an outright lie :)

Thanks for verifying this!

closed 2024-03-20 by txoof

txoof, 2024-03-20:
Updated documentation in in commit 4aa6912fd67265deeca859ed086ca3b637f73473