txoof/PaperPi #146: Support for the Inky Frame 7? (pico w built in)

petruut, 2024-02-07:


txoof, 2024-02-07:
I'm unfamiliar with this hardware. Can you provide more details?

txoof, 2024-02-09:
@petruut I think [this](https://shop.pimoroni.com/products/inky-frame-7-3?variant=40541882056787) is what you're talking about.

Unfortunately there's no easy way to port PaperPi to the Inky Frame without a major rewrite of all the supporting libraries for micro python. If there's a way to run the screen with pure python libraries, we could take a crack at it; it shouldn't bee terribly difficult.

That being said, it should be possible to port at least the core functionality of PaperPi to the inky frame using epdlib.  
 [epdlib](https://github.com/txoof/epdlib) lets you create layouts that will work on any 2D frame-buffered screen without having to "do the math" of where to put things. It automatically figures out how to scale and adjust layout elements. 

If you're interested, we could collaborate on doing this.

txoof, 2024-02-13:
@petruut 

I'm going to close this. If you have more to share, or want to work on porting this, let me know here.

closed 2024-02-13 by txoof