txoof/PaperPi #147: Daemon unable to start

OrangeTsar, 2024-02-08:
After installing, the daemon fails to run correctly,

**To Reproduce**
Follow installation instructions, run daemon

**Expected behavior**
A clear and concise description of what you expected to happen.
It's supposed to start

**Hardware**
- Full EPD version: epd7in5_V2
- Pi Version :Raspberry Pi 4 Model B Rev 1.5

**OS Info**
Run `cat /etc/os-release`
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

```

**`[main]` Section of Config**
Copy and paste the `[main]` section of your config file
- Daemon mode (start at boot) use: `/etc/default/paperpi.ini`
- Single user mode (start manually) use: `~/.config/com.txoof.paperpi/paperpi.ini`

```
[main]
# waveshare display type use HD for IT8951 displays
display_type = epd7in5_V2
# required vcom value for IT8951 screens
vcom = 0.0
# maximum refresh between total screen clear for HD displays
max_refresh = 4
# logging output level
log_level = WARNING
# display splash screen (True or False)
splash = True
# rotation of EPD display 
# (cable bottom: 0; cable left: -90; cable right: 90; cable top: 180)
rotation = 0
# mirror screen outuput - useful if output to screen appears backwards
mirror = False
# do not wipe screen when shutting down. True: leave image; False: wipe screen (default)
no_wipe = False
# plugin timeout - amount of time in seconds to wait for a hung plugin to complete execution
plugin_timeout = 30

```

I get the following when I run it:
```
musicbox systemd[1456]: paperpi-daemon.service: Failed to determine user credentials: No such process
musicbox systemd[1456]: paperpi-daemon.service: Failed at step USER spawning /usr/local/bin/paperpi: No such process
musicbox systemd[1]: Started PaperPi E-Paper Display.
musicbox systemd[1]: paperpi-daemon.service: Main process exited, code=exited, status=217/USER
musicbox systemd[1]: paperpi-daemon.service: Failed with result 'exit-code'.
musicbox systemd[1]: Started Session 4 of user jake.
musicbox systemd[1]: paperpi-daemon.service: Scheduled restart job, restart counter is at 4.
musicbox systemd[1]: Stopped PaperPi E-Paper Display.
```

txoof, 2024-02-08:
Thanks for the updated bug report.

I'll need to take a deeper look into this. I suspect something in the underlying LMS Query library is borked.

OrangeTsar, 2024-02-08:
It starts without that plugin loaded, and also the problem is intermittent. Sometimes it will start fine, sometimes it gives that error. If I keep trying to start it, it eventually goes.

txoof, 2024-02-09:
Are there any other log lines that might be relevant when it fails?

Good to know that the `paperpi --run_plugin_func lms_client.scan_servers` command also fails.

I'll dig into this to see if I can figure out what's going on.

txoof, 2024-02-09:
@OrangeTsar I've updated the QueryLMS library and paperpi. Can you try to reinstall paperpi and see if this fixes your issue?

OrangeTsar, 2024-02-10:
Now I get this when running paperpi with the LMS plugin and it doesn't output to the screen, I just get the fallback plugin
`QueryLMS:get_now_playing:742 :WARNING    - no valid playlist was returned`

OrangeTsar, 2024-02-10:
Appears to be working. It didn't like a LMS plugin that generated random playlists, but it's all good now it seems.

txoof, 2024-02-11:
@OrangeTsar glad to hear it! I'll close this ticket.

Not sure what you mean about the random playlists though. Can you elaborate?

closed 2024-02-11 by txoof