txoof/PaperPi #151: Does it work with waveshare 13.3 inch 1600x1200 ?

chopsuey5000, 2024-03-06:
I tried setting display_type as epd13in3k but it seems is for the 960×680 version. Is there any way of making it work with this model https://www.waveshare.com/13.3inch-e-Paper-HAT.htm ?

txoof, 2024-03-06:
## GOOD NEWS!
It looks like this is an IT8951 based board. This means you can take advantage of the partial refresh, fast update and high-resolution images.

**Use "HD" as the display type.**

I haven't directly tested a 13.3, I only have a 9" HD currently so I'd love to hear if you run into any problems. In theory, it should "just work." 

It also means that if you have a HiFiBerry attached, you can use it without any hassle.

txoof, 2024-03-08:
Thanks for the complete debugging info. That helps a lot.

Based on what I'm seeing in your output, it looks like the IT8951 library is failing to communicate with your board. I don't have a 13.3" to test with, but I think we can resolve this if you're willing to do some debugging. 

## Step 1 (the big easy solution)
Check that SPI is Enabled (the installer should have made you do this, but it's worth checking):

1. Run `sudo raspi-config` and check that `Interface Options > SPI` is set to "YES"
2. Reboot `sudo shutdown -r now` -- this isn't strictly necessary, but can't hurt
3. Try to start paperpi again and see where you get

## Step 2 A bit more hassle, but more data rich
We can also do some checking that your board is working properly:
1. change to the paperpi install location: `cd /usr/local/paperpi`
2. download IT8951 so we can use the test suite: `sgit clone --depth=1 https://github.com/GregDMeyer/IT8951.git`
3. start the virtual environment in the directory: `source ./venv_paperpi/bin/activate`
4. change to the test suite directory: `cd IT8951/test/integration/`
5. run the test `python3 ./test.py` - you should see a bunch of stuff show up on your screen and eventually a sleeping penguin
6. clear the screen and verity that everything is working: `python3 ./time_full.py`

Share the output from the testing.

If you're on Discord, you can visit the [support channel for PaperPi](https://discord.gg/K8Z7bfK5) and we can do some live debugging together too. I'm located in CET (UTC +1 Amsterdam/Berlin).

chopsuey5000, 2024-03-09:
Hi Aaron,

At the end i made a new fresh installation of Rpi OS + ParpePi and it worked at first like a charm. Previously, I was trying different ways of using the e-paper display with javascript or python and i think it created kind of mess up.

Thanks a lot for your answers.

Next step, develop a plugin ....

txoof, 2024-03-09:
So glad to hear its working!

If you want to collaborate on a plugin, let me know.

txoof, 2024-03-09:
I'll close this issue. If you want help with a plugin, open a new issue. 

PRs for plugins are always welcome.

closed 2024-03-09 by txoof