txoof/PaperPi #167: No module named 'ArgConfigParse'

daavid1995, 2025-02-09:
I tried to install PaperPi,
I got the message saying it was installed and I should configure it.
But then it never starts. The following message appears when I use the command "sudo /usr/local/bin/paperpi -d":

Traceback (most recent call last):
  File "/usr/local/paperpi/paperpi.py", line 32, in <module>
    import ArgConfigParse
ModuleNotFoundError: No module named 'ArgConfigParse'

Tried to install the modul with pip install ArgConfigParse, says installed and I see it with the command "pip list".

ArgConfigParse     0.2.7     /home/pi/.local/lib/python3.11/site-packages pip

Any idea?

Thanks

txoof, 2025-02-10:
It sounds like the virtual environment is failing to load for some reason. You've probably installed `ArgConfigParse` in your system python env. I need a little more info to diagnose the problem. 

Can you supply the following information?

**Hardware**
- Full EPD version: `paste here`
- Pi Version (use `cat /proc/device-tree/model`): `paste here`

**OS Info**
Run `cat /etc/os-release`
```
paste here
```

**`[main]` Section of Config**
Copy and paste the `[main]` section of your config file
- Daemon mode (start at boot) use: `/etc/default/paperpi.ini`
- Single user mode (start manually) use: `~/.config/com.txoof.paperpi/paperpi.ini`

```
paste here
```

Then try these steps and paste in the results

1. Run `/usr/local/bin/paperpi -V`:
2. run `cd /usr/local/paperpi; source /usr/local/paperpi/venv_paperpi/bin/activate`
3. run `pip list` and paste in the results below
4. and finally `ls -alh /usr/local/paperpi`

daavid1995, 2025-02-10:
Hardware
Full EPD version: epd2in13_v4 (2.13inch Touch e-Paper HAT)

Pi Version: Raspberry Pi Zero 2 W Rev 1.0pi

OS Info: PRETTY_NAME="Debian GNU/Linux 12 (bookworm)" NAME="Debian GNU/Linux" VERSION_ID="12" VERSION="12 (bookworm)" VERSION_CODENAME=bookworm ID=debian   

[main] Section of Config:  # CONFIG_VERSION=1.2 [main] # waveshare display type use HD for IT8951 displays display_type = epd2in13_V4 # required vcom value for IT8951 screens vcom = 0.0 # maximum refresh between total screen clear for HD > max_refresh = 4 # logging output level log_level = WARNING # display splash screen (True or False) splash = True # rotation of EPD display  # (cable bottom: 0; cable left: -90; cable right: 90> rotation = 0 # mirror screen outuput - useful if output to screen> mirror = False # Disable RGB color support for 7-color screens (Def> # This setting only affects 7 color screens and does> color = False # Disable screen wipe at shutdown (Default: True) # This setting will leave the last image on the scre> no_wipe = False # plugin timeout - amount of time in seconds to wait> plugin_timeout = 30 

/usr/local/bin/paperpi -V:
Traceback (most recent call last):   File "/usr/local/paperpi/paperpi.py", line 32, in <module>     import ArgConfigParse ModuleNotFoundError: No module named 'ArgConfigParse'   (venv_paperpi)

/usr/local/paperpi $ pip listPackage    Version ---------- ------- pip        23.0.1 setuptools 66.1.1

ls -alh /usr/local/paperpi  total 84K drwxr-xr-x  8 pi   pi   4.0K Feb  9 22:14 . drwxr-xr-x 11 root root 4.0K Feb  9 22:14 .. drwxr-xr-x  2 pi   pi   4.0K Feb  9 22:13 config -rw-r--r--  1 pi   pi    234 Feb  9 22:13 debian_packages-paperpi.txt drwxr-xr-x 15 pi   pi   4.0K Feb  9 22:13 fonts -rwxr-xr-x  1 root root  13K Feb  9 22:14 install.sh drwxr-xr-x  2 pi   pi   4.0K Feb  9 22:13 library -rw-r--r--  1 pi   pi    942 Feb  9 22:13 my_constants.py -rwxr-xr-x  1 pi   pi    27K Feb  9 22:13 paperpi.py drwxr-xr-x 18 pi   pi   4.0K Feb  9 22:13 plugins drwxr-xr-x  5 root root 4.0K Feb  9 22:14 venv_paperpi drwxr-xr-x  2 pi   pi   4.0K Feb  9 22:13 waveshare_epd

txoof, 2025-02-11:
Thanks for the info. 

Something is still really off are you sure you ran `cd /usr/local/paperpi; source /usr/local/paperpi/venv_paperpi/bin/activate; pip list`? It looks to me like you didn't launch the project venv before running pip.

If you're sure you did that and only got "setuptools", then something went horribly wrong with the install. There should be a huge list of things that were installed in the venv. 

The best option is to run the installer again. If you get similarly bad results with the installer, you can try this manual uninstall/install:

## DO THIS WITH CAUTION!

### Remove the old paperpi
1. `sudo rm /usr/local/bin/paperpi`
2. `sudo rm -rf /usr/local/paperpi`

### Manual Install PaperPi

1. `apt install git`
2. `cd ~/`
3. `git clone https://github.com/txoof/PaperPi.git`
4. `cd PaperPi`
5. `sudo ./install/install.sh`

If you see any errors, those would be super helpful for debugging the installer process. 

In the future, if you paste content from the terminal/log files/whatever, can you try using MarkDown code fences?  You can surround your pasted errors with three back tick characters "```" before and after the pasted content.

Code fences turns this: 
$ ls -alh
total 88K
drwxr-xr-x  9 pi   pi   4.0K Feb 10 17:19 .
drwxr-xr-x 12 root root 4.0K Sep 10 21:18 ..
drwxr-xr-x  2 pi   pi   4.0K Sep 10 21:16 config
-rw-r--r--  1 pi   pi    234 Sep 10 21:16 debian_packages-paperpi.txt
drwxr-xr-x 15 pi   pi   4.0K Sep 10 21:16 fonts

into this:
```
$ ls -alh
total 88K
drwxr-xr-x  9 pi   pi   4.0K Feb 10 17:19 .
drwxr-xr-x 12 root root 4.0K Sep 10 21:18 ..
drwxr-xr-x  2 pi   pi   4.0K Sep 10 21:16 config
-rw-r--r--  1 pi   pi    234 Sep 10 21:16 debian_packages-paperpi.txt
drwxr-xr-x 15 pi   pi   4.0K Sep 10 21:16 fonts
```

daavid1995, 2025-02-11:
Hi, thank you for your respons.

**Yes, I think I should be in the project venv:**

```pi@pizero:~ $ cd /usr/local/paperpi; source /usr/local/paperpi/venv_paperpi/bin/activate
(venv_paperpi) pi@pizero:/usr/local/paperpi $ pip list
Package    Version
---------- -------
pip        23.0.1
setuptools 66.1.1
(venv_paperpi) pi@pizero:/usr/local/paperpi $
```

**Manual Install PaperPi:**
```
pi@pizero:~/PaperPi $ sudo ./install/install.sh
checking if paperpi-daemon.service is running
paperpi-daemon.service not running
done
Checking permisisons
Permissions OK
Checking for required debian packages
found debian packages for PaperPi module debian_packages-paperpi.txt
checking libtiff-dev
checking libopenjp2-7
checking python3-pip
checking python3-venv
Required packages OK
Creating user: paperpi
Paperpi user alreay exists
Adding paperpi to the gpio and spi groups
Installing files to /usr/local/paperpi
Creating virtual environment in /usr/local/paperpi/venv_paperpi
Installing requirements from /tmp/tmp.lHMpKWqXfG
-i https://www.piwheels.org/simple/
--extra-index-url https://www.pypi.org/simple
argconfigparse
epdlib
git+https://github.com/GregDMeyer/IT8951.git
pillow
rpi.gpio
spidev

Pillow
requests

epdlib
QueryLMS
requests

dictor
epdlib
requests
dictor
pytz
requests
Pillow

feedparser
requests
Pillow
dictor
requests
psutil
dictor
Pillow
python_dateutil
requests
Looking in indexes: https://www.piwheels.org/simple/, https://www.pypi.org/simple
Collecting git+https://github.com/GregDMeyer/IT8951.git (from -r /tmp/tmp.lHMpKWqXfG (line 5))
[... 146 lines cut ...]

you selected to run on demand
you can enable the daemon later by typing:
$ sudo systemctl enable paperpi-daemon.service

checking if SPI is enabled

    You must now complete the following steps
    REQUIRED:
    * edit /etc/default/paperpi.ini and set:
      - display_type = [YOUR_SCREEN]
      - vcom = [only set for HD screens]

    OPTIONAL:
    * Enable plugins by removing the "x" from section headers
    * Configure the plugins to match your needs/environment

Would you like to do that now? (y/N): y

install completed

```

Hope it helps somehow!
BR, David

txoof, 2025-02-20:
@daavid1995  Sorry I missed this. I've been on the road. I'll try to take a look over the weekend.

AaronCiuffo245484, 2025-02-20:
The IT8951 error is pretty clearly a problem and looks like it might be causing the entire install to bail out. You don't need the module that's failing, so there should be a work around. I'll see if I can find some time this weekend to fix this.

closed 2026-11-09 by txoof-bot