txoof/PaperPi #170: cannot import name '_imagingft' from 'PIL'

williemctell, 2025-05-06:
When starting paperpi via systemctl-- sudo systemctl paperpi-daemon.service paperpi fails to start due to the error "cannot import name '_imagingft' from 'PIL'"

> May 06 10:24:24 2W-e-ink paperpi[3004]:   File "/usr/local/paperpi/venv_paperpi/lib/python3.11/site-packages/PIL/ImageFont.py", line 877, in freetype
May 06 10:24:24 2W-e-ink paperpi[3004]:     return FreeTypeFont(font, size, index, encoding, layout_engine)
May 06 10:24:24 2W-e-ink paperpi[3004]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
May 06 10:24:24 2W-e-ink paperpi[3004]:   File "/usr/local/paperpi/venv_paperpi/lib/python3.11/site-packages/PIL/ImageFont.py", line 228, in __init__
May 06 10:24:24 2W-e-ink paperpi[3004]:     raise core.ex
May 06 10:24:24 2W-e-ink paperpi[3004]:   File "/usr/local/paperpi/venv_paperpi/lib/python3.11/site-packages/PIL/ImageFont.py", line 67, in <module>
May 06 10:24:24 2W-e-ink paperpi[3004]:     from . import _imagingft as core
May 06 10:24:24 2W-e-ink paperpi[3004]: ImportError: cannot import name '_imagingft' from 'PIL' (/usr/local/paperpi/venv_paperpi/lib/python3.11/site-packages/PIL/__init__.py)
May 06 10:24:25 2W-e-ink systemd[1]: paperpi-daemon.service: Deactivated successfully.
May 06 10:24:25 2W-e-ink systemd[1]: paperpi-daemon.service: Consumed 11.095s CPU time.
 

To reproduce enter "sudo systemctl start paperpi-daemon.service" on CL
*
- Full EPD version:  epd4in2b_V2
- Pi Version (use `cat /proc/device-tree/model`):  Zero W Rev 1

**OS Info**
Run `cat /etc/os-release`
```
Raspbian GNU/Linux 12 (bookworm)
```

**`[main]` Section of Config**
Copy and paste the `[main]` section of your config file
- Daemon mode (start at boot) use: `/etc/default/paperpi.ini`
- Single user mode (start manually) use: `~/.config/com.txoof.paperpi/paperpi.ini`

```
# CONFIG_VERSION=1.2
[main]
# waveshare display type use HD for IT8951 displays
display_type = epd4in2b_V2
# required vcom value for IT8951 screens
vcom = 0.0
# maximum refresh between total screen clear for HD displays
max_refresh = 4
# logging output level
log_level = WARNING
# display splash screen (True or False)
splash = True
# rotation of EPD display
# (cable bottom: 0; cable left: -90; cable right: 90; cable top: 180)
rotation = 0
# mirror screen outuput - useful if output to screen appears backwards
mirror = False
# Disable RGB color support for 7-color screens (Default: True)
# This setting only affects 7 color screens and does not impact bi-color screens
color = False
# Disable screen wipe at shutdown (Default: True)
# This setting will leave the last image on the screen at shutdown
no_wipe = False
# plugin timeout - amount of time in seconds to wait for a hung plugin to complete execution
plugin_timeout = 30

```

txoof, 2025-05-07:
Thanks for the complete ticket.

Does this happen when you launch it both in daemon and in single-user mode?  Can you try running paperpi manually from the command line as a regular user?

It looks like something failed to install with Pillow. I have idea what the problem is, but I need a little more information.

williemctell, 2025-05-07:
Here's the system log for starting as a daemon with the debug plugin turned on:  

May 07 10:10:20 2W-e-ink sudo[2266]:    admin : TTY=pts/0 ; PWD=/home/admin ; US                                                                               ER=root ; COMMAND=/usr/bin/systemctl restart paperpi-daemon.service
May 07 10:10:20 2W-e-ink sudo[2266]: pam_unix(sudo:session): session opened for                                                                                user root(uid=0) by admin(uid=1000)
May 07 10:10:21 2W-e-ink systemd[1]: Started paperpi-daemon.service - PaperPi E-                                                                               Paper Display.
May 07 10:10:21 2W-e-ink sudo[2266]: pam_unix(sudo:session): session closed for                                                                                user root
May 07 10:10:25 2W-e-ink paperpi[2272]: Traceback (most recent call last):
May 07 10:10:25 2W-e-ink paperpi[2272]:   File "/usr/local/paperpi/paperpi.py",                                                                                line 32, in <module>
May 07 10:10:25 2W-e-ink paperpi[2272]:     import ArgConfigParse
May 07 10:10:25 2W-e-ink paperpi[2272]: ModuleNotFoundError: No module named 'Ar                                                                               gConfigParse'
May 07 10:10:26 2W-e-ink systemd[1]: paperpi-daemon.service: Deactivated success                                                                               fully.
May 07 10:10:26 2W-e-ink systemd[1]: paperpi-daemon.service: Consumed 5.121s CPU                                                                                time.

The same error is thrown when I start in single user mode from the command line.

sudo /usr/local/bin/paperpi -d
Traceback (most recent call last):
  File "/usr/local/paperpi/paperpi.py", line 32, in <module>
    import ArgConfigParse
ModuleNotFoundError: No module named 'ArgConfigParse'

txoof, 2025-05-08:
I see some really strange things in the logs. In the second line, the module name has a space in it.

```
May 07 10:10:25 2W-e-ink paperpi[2272]: import ArgConfigParse
May 07 10:10:25 2W-e-ink paperpi[2272]: ModuleNotFoundError: No module named 'Ar gConfigParse'
```

Is that a copy/paste error, or is that actually showing up? There should be no space in the module name. I'm looking in the source and I can't find anything like that.

If it is just a copy/paste error, we can ignore it.

Try running  `$ /usr/local/bin/paperpi -l DEBUG`. Don't run it as sudo and see what the logs say. If you get the same errors, we know that the environment is totally broken. At that point, try reinstalling from the source.

If it works and dies at the PIL module error  again, there's something else at work. Try this:

```
$ cd /usr/local/paperpi/ 
$ source ./venv_paperpi/bin/activate
$ which python
/usr/local/paperpi/venv_paperpi/bin/python
```

This should activate the virtual environment the program runs in so we can run a test. If you don't see `/usr/local/paperpi/venv_paperpi/bin/python` when you run `which`, then we know the environment is messed up.

Finally try this:

```
python << EOF
import PIL
import ArgConfigParse
from PIL import _imagingft as i
print(help(i))
EOF
```

If that fails spectacularly, the environment is likely busted and we need to try a reinstall.

williemctell, 2025-05-08:
The space in ArgConfig must have been a typo.  Running paperpi from the command line with the DEBUG parms resulted in the ArgConfigParse error.  which gave the expected result.  python << EOF... threw the error  No module named 'PIL'.

I'll try uninstalling followed by a reinstall.

AaronCiuffo245484, 2025-05-09:
That's super helpful. It looks like there's something wrong with the installer. I'll see if I can get there this weekend.

txoof, 2025-05-10:
@williemctell I'm wondering if this has something to do with the fact that the W identifies as an `arm6l`. I suspect that some dependency or another goes off the rails when it tries to build against cython for arm6l. It wouldn't surprise me if this causes all sorts of issues later down the pipeline.

I can't replicate this on a pi4 with a freshly built installed and upgrade Raspi 64 Lite. I've just spun up a new pi image and run through the remote-install script and It's running as expected with no crashes and a valid environment. 

Do you have a pi3 or 4 that you can try this on? I don't have any Zeros to debug on here and investigate deeper.

williemctell, 2025-05-11:
I got it started on a 3A+.  An oddity was that the version that worked for me in the ini file for the 4.2 spi screen was epd4in2_V2 which is deprecated in the list of screen versions.  I tried the screen version on the 2W and got the same error as before.  I'm a retired programmer and remember the amount of time I spent debugging my work vs the conceptual and coding time.  :)

williemctell, 2025-05-12:
Go ahead and close the ticket.  I appreciate your help.

closed 2025-05-13 by txoof

txoof, 2025-06-22:
@jasonn514 The Pi zero is not currently supported because it uses 32 bit versions of a lot of libraries. When the python libraries get installed, this causes problems that cannot be easily resolved.

A raspberry PI 3 will run paperpi and a 64 bit version of debian just fine if you can get your hands on one of those.

jasonn514, 2025-06-22:
Ok, thanks for the feedback.

txoof, 2025-10-19:
@alperinan can you provide a link to where you see it supported? The last time I checked, pi zero only supports 32 bit versions of Raspbian and that is incompatible with some of the libraries paper pi relies on.

alperinan, 2025-10-19:
> [@alperinan](https://github.com/alperinan) ahh. Found it. That's an error. I'll need to fix that. Or, provide a PR) if you have a second.

oh no worries. I am actually very new to raspbery world, I am more a traditional software engineer but your project is very interesting so I want to learn. Another thing I am trying to understand your project requires 32 or 64 bit version ? I see zero W has both but I did not which one to use so I was using 32 bit version.