txoof/PaperPi #173: 4inch HAT+ (E) not supported

dedenker, 2025-06-02:
**Describe the bug**
I have a Waveshare model that is not working with current supported models.
This is the [wiki page](https://www.waveshare.com/wiki/4inch_e-Paper_HAT%2B_(E)_Manual), where I can also see it  is closest to the **epd4in01f** model.
But mine is full color and 600x400 (instead of 640x400).

**To Reproduce**
Trying any of the 4inch modules results in nothing on display, but also no error.

**Expected behavior**
Working display

**OS Info**
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
Raspberry zero W
```

**Additional context**
Running the example from Waveshare works fine.
Somehow current PaperPi doesnt support this display.

txoof, 2025-06-04:
Thanks for the complete ticket. I think the problem is that you are running in a Zero. We discovered recently that there are some libraries that go off the rails when built on the Zero. 

Can you try on a pi3/4?

dedenker, 2025-06-05:
The first test was with bookworm (32bit), which gives the issue with **ImportError: cannot import name '_imagingft' from 'PIL'**.
So re-imaged with bulleyes (my zero is imaged with this) and now get
` PinFactoryFallback: Falling back from lgpio: /lib/arm-linux-gnueabihf/libc.so.6: version GLIBC_2.33 not found` which also accured with the RasPi zero.
This I could fix with updating the virtual environment of PaperPi, but finally had the same as the RasPi Zero, no error but also no display output.

But! I look again at the Waveshare [example/demo](https://files.waveshare.com/wiki/4inch-e-Paper-HAT%2B-(E)/4inch_e-Paper_E.zip) and I could 1on1 copy that library and that works!
Not a single line of code done by me...So feel weird to commit this.

Another issue I see is that colors dont come up.
But that is not for this issue# and I will investigate this first more before opening a new one.

txoof, 2025-06-13:
Bookworm 32 will likely always fail for one reason or another. Everything is built against static libs in bookworm 64.

If you can somehow get a 64 but version running on a zero, you are good to go.

The reason the Wave share demo works is it is just pushing static images. Paperpi is doing a whole lot more.

Open the waveshare demo and take a look around. Then open paperpi and look at all the imports. 

Did you get something to run?