txoof/PaperPi #174: No display and getting padded area too small warning

raspberryshromp, 2025-06-05:
**Describe the bug**
Not getting any display. getting the warning: 

This is the error I got last install

15:19:21 Block :padding :340 :WARNING - the padded area available may be too small to display any content: Area: (360, 24), Padded Area: [340, 4]
15:19:53 Block :padding :340 :WARNING - the padded area available may be too small to display any content: Area: (216, 24), Padded Area: [196, 4]
15:19:57 Block :padding :340 :WARNING - the padded area available may be too small to display any content: Area: (144, 24), Padded Area: [124, 4]

Getting this same warning after fresh install and going through the step by step install again still no display output..
07:10:36 Block   :padding   :340 :WARNING    - the padded area available may be too small to display any content: Area: (360, 24), Padded Area: [340, 4]

**To Reproduce**
I don't know how to reproduce as I have tried again from a fresh install and just followed the step by step install guide. 

**Expected behavior**
Screen to display plugins as they should. 

**Hardware**
- Full EPD version: epd3in52
- Pi Version Raspberry Pi 3 Model B Rev 1.2

**OS Info**
```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

**`[main]` Section of Config**
# CONFIG_VERSION=1.2
[main]
# waveshare display type use HD for IT8951 displays
display_type = epd3in52
# required vcom value for IT8951 screens
vcom = 0.0
# maximum refresh between total screen clear for HD displays
max_refresh = 4
# logging output level
log_level = WARNING
# display splash screen (True or False)
splash = True
# rotation of EPD display
# (cable bottom: 0; cable left: -90; cable right: 90; cable top: 180)
rotation = 0
# mirror screen outuput - useful if output to screen appears backwards
mirror = False
# Disable RGB color support for 7-color screens (Default: True)
# This setting only affects 7 color screens and does not impact bi-color screens
color = false
# Disable screen wipe at shutdown (Default: True)
# This setting will leave the last image on the screen at shutdown
no_wipe = False
# plugin timeout - amount of time in seconds to wait for a hung plugin to complete execution
plugin_timeout = 30

```

```

**Plugin Issues**
The these are the only two plugins enabled

```
[Plugin: default fallback plugin]
layout = layout
plugin = default
refresh_rate = 30
min_display_time = 60
max_priority = 2**15

[Plugin: Basic Clock]
layout = layout
plugin = basic_clock
refresh_rate = 30
min_display_time = 50
max_priority = 2
```

**Additional context**
Initially I was trying the weather plugin but to troubleshoot I switched to the basic clock one since its simpler to see if maybe the other plugin had too much and the screen was not able to display it properly. Display does work fine as I have tested it using the waveshare demos.

txoof, 2025-06-05:
Thanks for the ticket. This helps.

Switch set the logging to `INFO` and disable the fallback plugin - `[#Plugin: default fallback plugin]`

Run it from the command line, not in daemon mode. We're looking for any of the plugin update info. We want to know what plugin is refreshing and what blocks it is rendering.

The basic clock should definitely work on your display.

txoof, 2025-06-13:
Deep in some time sensitive projects. Sorry my replays are slow.

Just run with `paperpi` from the command line I assume you're sharing the `~/com.txoof.paperpi/paperpi.ini`? If not try `paperpi -c /etc/defaults/paperpi.ini`

I'm working from memory on the train so I might have fumbled the paths. Use the appropriate path for your configuration.

closed 2026-11-09 by txoof-bot